mcu-tools/mcuboot #495: sector erase is not limited to erasing sectors

solsbarry, 2019-06-26:
I was having an issue that my REVERT swap was failing. I noticed that it was because during the swap, only the first sector of my secondary slot was being erased before the primary slot image was copied over. This was because in loader.c the swap function was doing "boot_erase_sector" with a "sz" parameter larger than a sector. My flash driver only accepts erase commands where the size is one sector. I fixed this by putting a for loop in the boot_erase_sector function. 

```c
static inline int
boot_erase_sector(const struct flash_area *fap, uint32_t off, uint32_t sz)
{
    int i, rc;
    int sector_sz = (*boot_data.imgs[BOOT_PRIMARY_SLOT].sectors).fs_size;
    int num_sectors = sz / sector_sz  + (sz % sector_sz > 0 ? 1 : 0);
    for (i=0; i < num_sectors; i++){
      rc = flash_area_erase(fap, off+sector_sz*i, sector_sz);
      if (rc != 0) return rc;
    }
    return rc;
}
```

I am not sure if in the MCUboot design ethos, you would rather this change be handled in the driver.  But I would point out that the name of the function here is "boot_erase_sector" not "boot_erase_region"

utzig, 2019-06-26:
Are you working on a new port?

The erase operation is delegated to `flash_area_erase` which should handle it, be it one sector on any other amount, so it would be better for you to leave the function unchanged and move the logic there. The naming suggestion makes sense!

solsbarry, 2019-06-26:
Not a new port. I am using Zephyr on an NRF52840. You are correct that the flash_area_erase should be able to handle any size. In my case I am erasing my external flash device (where my secondary slot is) and flash_area_erase maps eventually to my qspi flash driver, which I guess is where I should make the change so that it can support erase requests of any size. My qspi flash driver only accepts sector erases right now.

closed 2019-06-26 by solsbarry

nvlsianpu, 2019-06-28:
flash_area_erase() supports erase at size of multiples flash pages in fact,
Good to saw that somebody want to use an external memory for the slot.