mcu-tools/mcuboot #506: Dependencies in TLV are insecure

d3zd3z, 2019-07-12:
The way dependency information is encoded in the TLV is not secure. Prior to the multi-image change adding dependency information, the TLV was entirely concerned with the hash an signature of the image itself. As a consequence, tampering with the TLV would just result in the signature or hash being invalid, and as long as the code validating the TLV is correct, this prevents tampering.

However, adding a TLV entry for dependencies now introduces a TLV entry that is not protected. It becomes trivial to tamper with this TLV entry, allowing upgrades to be done without satisfying the proper dependencies.

I propose the following changes to the TLV format and layout, which we can implement. This is a breaking change, however the code could support the old version as long as there are no dependencies present.

1. Enhance the ordering requirements on the TLV entries. Currently, the code doesn't really check the order, although the dependency code assumes the dependency TLVs are all grouped together. I propose the following order requirement: TLV_SHA256 first, followed by any of the TLV_ENC entries, then any TLV_DEPENDENCY entries. After this, starting with the first TLV_KEYHASH item will only be pairs of TLV_KEYHASH and one of the signature entries.

2. Change the semantics of the signatures. Instead of signing the value of the TLV_SHA256, instead the signature covers the block of data including the `image_tlv_info` header, and all of the TLV entries themselves, up until the first TLV_KEYHASH entry. These will be done using the same existing signature algorithms, which will involve computing a SHA256 of this region of data, and signing that hash.

There are a few ways of handling the semantic difference between the old TLV format and the new:

1. Change the TLV `it_magic` value to distinguish the new format of TLV.
2. Retire all of the TLV_RSA2048_PSS TLV values, and create new ones for the signature over the TLV instead of signature of the image.
3. Introduce a "marker" TLV that goes between the last dependency (or SHA256 or KEY) and the first TLV_KEYHASH. Something like `IMAGE_TLV_END_SIGNED_DATA`. The presence of this would indicate that the following signatures are of the data up unitl the end-signed-data marker. If there is a signature without this marker, it indicates a signature of the image instead.

I propose the third alternative.

Another alternative is to use [COSE](https://tools.ietf.org/html/rfc8152), which would homogenize with future [SUIT](https://datatracker.ietf.org/wg/suit/about/) support, but likely introduce quite a bit more code.

d3zd3z, 2019-07-12:
There is no need to embargo this vulnerability, as no release has been made with this code (in fact, as of this writing, the change has not even been merged). However, this should block a release once the multi-image code is merged, as it introduces an insecure way of encoding dependencies.

tamasban, 2019-07-15:
The dependency information in TLV is covered by the signature. The border of signature  coverage was extended in #369:
- One of the padding bytes in image header was converted to `ih_protect_tlv_size`. The size  of TLV header is also included here. 
- Those TLVs which must be signature protected, they are coming first after the image:
   - Header + image + TLV header + Protected TLVs (currently only dependency TLV) + unprotected TLVs (all other)
- `imgtool` and image validation code is updated to act accordingly

d3zd3z, 2019-08-10:
I'm going to close this issue, as it describes a vulnerability that the code at least tries to protect against.

closed 2019-08-10 by d3zd3z