mcu-tools/mcuboot #508: Document generating encrypted images with imgtool

AGlass0fMilk, 2019-07-19:
Generating an encrypted update candidate is not very clear from the imgtool documentation at the moment.

Why does the source code have a built-in encryption key (at least for zephyr)? Here:

https://github.com/JuulLabs-OSS/mcuboot/blob/853657c23d385fd105b72833c8c0cfb2a82a45a2/boot/zephyr/keys.c#L72-L171

Are any bytes of these key randomized as the documentation says? What about the RSA key passed to imgtool with the `-E` flag? Where is that supposed to be put in the bootloader code? Built into the keys array? Can it be the same key used to sign update candidates?

utzig, 2019-07-19:
The key in `keys.c` for Zephyr/Mynewt/Simulator corresponds to the `enc-rsa2048-priv.pem` file in the root directory. This is just for sample tests. `enc_priv_key[]` in keys.c needs to be updated with a dump of your generated keypair. I will add support for auto-generation, like it's currently done for signing keys. The main idea was to have the decryption key to live somewhere else in the MCU, not in flash, like in a "protected" internal ROM, that one cannot dump, but there is no standard API for this (so it would not be in flash and not be part of the source code like today).

> Can it be the same key used to sign update candidates?

No, never ever reuse those keys! Inside the bootloader there is a verification key which is public and protecting the corresponding signing key should be the most important security measure to take. For encryption the public key is used to encrypt and the private (inside the bootloader) to decrypt.

AGlass0fMilk, 2019-07-19:
@utzig Thanks for your response. What kind of key is the enc_priv_key then? If it's RSA-2048, why is it 1192 bytes long?

utzig, 2019-07-19:
It is a hexdump of the DER encoding of the keypair. The 2048 means just the size of the modulus, you could, for example, extract the public key which consists of a modulus and a public exponent and you would be able to verify that the modulus is actually 2048 bits long.

closed 2021-06-03 by github-actions[bot]