mcu-tools/mcuboot #518: Multi-image upgrade with non-satisfied dependencies hangs boot forever

utzig, 2019-08-05:
When dependency check is done, if the version available is less than the expected version, the checking routines hangs forever. The correct thing to do would be to boot the current version and not try to upgrade.

See: https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/bootutil/src/loader.c#L1792

tamasban, 2019-08-27:
Could you elaborate when can this happen:
- The dependencies of original images are satisfied?
- How many image is tried to be upgraded? 1 or 2?
- I assume the new image dependency is not satisfied.
- Does the dependency check restarts forever?

d3zd3z, 2019-12-02:
@utzig is this fixed now?

utzig, 2019-12-03:
Yes, was fixed at the same time that dependency testing was added to the simulator.

closed 2019-12-03 by utzig