mcu-tools/mcuboot #522: Non page-aligned flash areas gets incorrect num_sectors

hakonfam, 2019-08-07:
TL;DR: Is it required that flash regions are page aligned?

AFAICS, all flash regions must be flash page aligned. This because of how the number of sectors is calculated.

```
 98         for (i = 0; i < l->pages_count; i++) {
 99             page_info.start_offset = off;
100             page_info.index = page;
101 
102             if (!cb(&page_info, data)) {
103                 return;
104             }
```

Hence, having following flash layout will not work, you can see that the num_sectors is different (0x5f and 0x5e)
```
>>> p boot_data.imgs[0].num_sectors 
$67 = 0x5f
>>> p *boot_data.imgs[0].area
$68 = {
  fa_id = 0x5, 
  fa_device_id = 0x0, 
  pad16 = 0x0, 
  fa_off = 0x20000, 
  fa_size = 0x5e800, 
  fa_dev_name = 0x10148 "NRF_FLASH_DRV_NAME"
}
>>> p boot_data.imgs[1].num_sectors 
$69 = 0x5e
>>> p *boot_data.imgs[1].area
$70 = {
  fa_id = 0x8, 
  fa_device_id = 0x0, 
  pad16 = 0x0, 
  fa_off = 0x7e800, 
  fa_size = 0x5e800, 
  fa_dev_name = 0x10148 "NRF_FLASH_DRV_NAME"
}
>>>
```

Is it required to have page aligned memory regions? If yes, should I create a PR which checks this and gives a detailed error message? If no, should it be documented that this is required, or should the code be fixed to work with non-page-aligned flash areas.

nvlsianpu, 2019-08-07:
It is assumption that mcuboot areas are page aligned. (for anyone page size from case above is 0x1000 B). You are right that error log should better covered this case.

Patch for better error message will be welcome of course.

utzig, 2019-08-07:
How does one detect if a page is aligned? Is this information available in a OS and arch independent fashion?

Laczen, 2019-08-09:
@hakonfam, areas and sectors must be page aligned to allow an erase. If areas and sectors are not page aligned not only will you get incorrect number of sectors all erases will fail. So it's not that sectors need to be page aligned to be the correct number of sectors, it's the number of sectors can.be is calculated correct because sectors and areas are page aligned.

d3zd3z, 2019-12-02:
I believe this has been answered. Feel free to submit a pull request with documentation changes if anyone feels that would be helpful.

closed 2019-12-02 by d3zd3z