mcu-tools/mcuboot #529: MCUBoot not cleaning up properly before booting Zephyr?

endian-benjamin, 2019-08-14:
Cross post from: https://github.com/zephyrproject-rtos/zephyr/issues/18231

If you enable UART or RTT logging in MCUBoot, it seems these won't be uninitialized after booting to Zephyr, at least on nRF52.

Steps to reproduce the behavior:
1. Build a sample for nRF52 that implements proper power management (suspending all UARTs etc) and flash it "as usual" i.e. at offset `0x00` with bootloader disabled
2. Observe that power consumption is expectedly low when UARTs are disabled, about 4 uA hopefully
3. Build/flash mcuboot with UART enabled and redo step 1, flashing at `slot-0`
4. Observe that power consumption of `slot-0` application behaves as if UART never gets suspended.

Expected behavior:
- `slot-0` application behaves the same wrt power consumption regardless if you're bootloading or not.

Impact:
- If one does not rely upon UART inside bootloader, simply disabling it in the mcuboot `prj.conf` seems like a fine workaround. But I figured it was worth reporting anyway in case this is an indication that far worse bugs could be possible.

nvlsianpu, 2019-08-14:
Thanks for report it here!

utzig, 2019-08-14:
@mbolivar You were looking into something to handle this, correct?

mbolivar, 2019-08-17:
> @mbolivar You were looking into something to handle this, correct?

We briefly discussed this at an MCUboot meeting and decided Zephyr needs better power management for disabling devices. I'm not actively working on that but it would be the next step.

endian-benjamin, 2019-08-19:
So basically the issue is that we're not uniniting because Zephyr lacks "uninit" APIs?

mbolivar, 2019-08-19:
> So basically the issue is that we're not uniniting because Zephyr lacks "uninit" APIs?

Yep. I think @pabigot is starting to work on device power management but I am not sure.

pabigot, 2019-08-19:
I am, yes.  I expect it to take a while.  One might say that Zephyr's got some architectural weaknesses in areas that are critical to power conservation.

endian-benjamin, 2019-08-20:
@pabigot I am well aware :P I'll pm you on slack, since power management is dear to my heart.

endian-benjamin, 2020-02-06:
@pabigot Situation unchanged?

pabigot, 2020-02-06:
Yes.

nvlsianpu, 2020-02-06:
I think its time to add cleanup functionality to mcuboot, at last temporary.

pabigot, 2020-02-06:
> I think its time to add cleanup functionality to mcuboot, at last temporary.

That would probably be wise.  By "unchanged" I mean there's strong interest in talking about how we're going to do this (the Zephyr dev meeting is starting to focus on some devicetree improvements that would help enable the feature).  But nobody has a clear plan and committed resources, and until that happens it's all smiles and visions of a wonderful future.

closed 2021-06-03 by github-actions[bot]