mcu-tools/mcuboot #530: HARD FAULT writing or reading flash on a stm32f0

lucaspeixotot, 2019-08-14:
I'm trying to swap images from slot1 to slot0 with boot_request_upgrade(0) and I'm having problems. When the mcuboot try to manipulate some region of flash like slot0, slot1 or scratch through flash_area_write or flash_area_erase I get this error:

```
***** HARD FAULT *****
***** Hardware exception *****
Current thread ID = 0x00000000
Faulting instruction address = 0xfffffffe
```

More precisely the error is happening after to call this function flash_write_protection_set(flash_dev, false).

I was wondering if my problem was that I couldn't access flash for some reason, so I erased my memory flash and flashed a code with these statements and no errors occurred.

```
u8_t buff[] = "aaaaaaa";
int err            = flash_area_open(SCRATCH_ID, &fap);
if (!err) {
    printk("scratch -> fa_off: %X, fa_size: %X, fa_id: %d, fa_device_id: %d\n", fap->fa_off,
           fap->fa_size, fap->fa_id, fap->fa_device_id);
}
err = flash_area_write(fap, 0, buff, strlen(buff));
if (err) {
    printk("Error writing...\n");
} 
err = flash_area_erase(fap, 0, fap->fa_size);
if (err) {
    printk("Error erasing...\n");
}
```

After that, I got this same code and put it in the begin of mcuboot's main and I got HARD FAULT again. So the same code results in HARD FAULT when I use it in mcuboot code and results in no error when I use it in a blank project.

I'm using the code in the branch master.

nvlsianpu, 2019-08-14:
Are you able to run zephyr flash_map test on your board? 
https://github.com/zephyrproject-rtos/zephyr/tree/master/tests/subsys/storage/flash_map

lucaspeixotot, 2019-08-14:
Yes, everything is ok with this test.

lucaspeixotot, 2019-08-14:
I tried to use mcuboot with a nrf board and everything works fine, the problem is just with a stm32f0 board.

utzig, 2019-08-14:
What particular board are you using?

lucaspeixotot, 2019-08-14:
> What particular board are you using?

nucleo_f091rc

lucaspeixotot, 2019-08-15:
Today I realized that the hard fault occurs when the semaphore in flash_write_protection_set function is being given. What is curious is that this driver works properly in other projects than are not mcuboot.

utzig, 2019-08-15:
I am not a Zephyr developer, but MCUBoot runs without `CONFIG_MULTITHREADING`. The STM32 flash driver is using semaphores, and looking at the code I see `z_impl_k_sem_give` actually calls `z_reschedule`. I have no idea how that works when multi-threading is disabled (if it does at all). Also the nRF driver #ifdefs out the semaphores when no multi-threading is available: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/flash/soc_flash_nrf.c#L69

So maybe one test you could do would be to define `CONFIG_MULTITHREADING` for testing purposes. Anyway I never used STM32 with Zephyr but if the flash driver is broken, it is broken for every STM32; it would be surprising, so I am probably wrong here!

lucaspeixotot, 2019-08-15:
> I am not a Zephyr developer, but MCUBoot runs without `CONFIG_MULTITHREADING`. The STM32 flash driver is using semaphores, and looking at the code I see `z_impl_k_sem_give` actually calls [...]

The problem was exactly this! I enabled CONFIG_MULTITHREADING and it worked right away.
Thanks for your help @utzig!

I'm closing this issue since the problem is solved.

utzig, 2019-08-15:
Thanks @lucaspeixotot, I will leave this open because one is not supposed to enable `CONFIG_MULTITHREADING` for a bootloader!

@carlescufi @mbolivar @nvlsianpu Could you see moving this into a Zephyr issue?

mbolivar, 2019-08-17:
As far as I know, `CONFIG_MULTITHREADING` is semi-deprecated in Zephyr, and various drivers are expected to break if it is disabled. A mess, I know, but that's the state of affairs.

nvlsianpu, 2019-08-19:
@mbolivar 

> As far as I know, CONFIG_MULTITHREADING is semi-deprecated in Zephyr, and various drivers are expected to break if it is disabled. A mess, I know, but that's the state of affairs.

Actually I and a few other people pressed to keep possibility of disabling multithreading in zephyr. This helps to decrease flash footprint significantly. I proposed to make a statement about how driver should support mcuboot in order to operate without multithreading (proposal was ignored). For instance nRF  flash driver doesn't use kernel synchronization primitives when compiled for no-multithreading, and uses them when multithreading is enabled.

mbolivar, 2019-08-19:
@nvlsianpu OK, in that case, I wonder if we need to add some extra Kconfig options to the flash driver to signal that it is safe to use with MULTITHREADING=n, so the user can get a failed build if it isn't? This is going to keep happening over and over without it.

mbolivar, 2019-08-19:
BTW, is Andy Ross on board with this idea?

utzig, 2019-08-19:
Is this a temporary solution? Because it seems to me that things should either work for some specific config or make it clear that they don't. It should not be the driver itself having to find out, but the semaphore code, no?

nvlsianpu, 2019-08-19:
@mbolivar 

> OK, in that case, I wonder if we need to add some extra Kconfig options to the flash driver to signal that it is safe to use with MULTITHREADING=n, so the user can get a failed build if it isn't? This is going to keep happening over and over without it.

What about processor check for each flash driver which requires multithreading?
Be aware that mcuboot might potentially use a few flash driver on the same SoC.

mbolivar, 2019-08-19:
```
add_library(${KERNEL_LIBRARY}
  device.c
  errno.c
  fatal.c
  idle.c
  init.c
  mailbox.c
  mem_slab.c
  mempool.c
  msg_q.c
  mutex.c
  pipes.c
  queue.c
  sched.c
  sem.c
  stack.c
  system_work_q.c
  thread.c
  thread_abort.c
  version.c
  work_q.c
  smp.c
  )
```

On second thought, I think we should convert this line so that these files are only built using `target_sources_ifdef(CONFIG_MULTITHREADING ...)` or so as appropriate (e.g. for sem.c).

Otherwise this is just going to continue to break randomly -- which is why I wonder how Andy Ross feels about this.

nvlsianpu, 2019-08-19:
@mbolivar I like the idea you showed. I believe worth to open a zephyr PR with this patch inside.

d3zd3z, 2019-08-19:
At least one of the flash drivers I looked at only needed synchronization for mutual exclusion. Unfortunately, mutexes are a posix feature, not a core kernel feature. It would be easy to build mutex code that when configured single threaded just implemented an interrupt lock. But, you can’t do this with semaphores, as they are too general, and the code can’t tell how they are being used.

mbolivar, 2019-08-19:
@nvlsianpu I've asked some of the kernel devs what they think about this.

closed 2021-06-03 by github-actions[bot]