mcu-tools/mcuboot #531: Add Zephyr builds to CI

utzig, 2019-08-16:
Currently the bootloader is tested on a simulator, and a multiple target configurations are built for Mynewt. Similar building tests must be added for Zephyr to detect build failures.

d3zd3z, 2019-12-03:
Any thoughts on pulling in the Zephyr SDK? It is fairly large, and takes some time to install.

nvlsianpu, 2019-12-05:
Zephyr SDK is a GCC compilation in fact.
For NCS we are using  GNU Arm Embedded Toolchain  8-2019-q3- (http://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.1.0/nrf/gs_ins_windows.html#installing-the-toolchain)
I know that this helps with limited amount of arm architectures, but maybe will be enough.

utzig, 2019-12-05:
For Mynewt we use the same toolchain: https://github.com/JuulLabs-OSS/mcuboot/blob/master/ci/mynewt_install.sh#L39

The point is not to support different architectures, but support different MCUBoot conffigurations, even for a single target (eg, nrf52840_pca10056 or frdm_k64f).

d3zd3z, 2019-12-05:
I do read that Zephyr can build with the individual tools instead of the SDK. I guess at that point, it is a matter of figuring out what needs to be installed. I'll play with it a bit at some point.

closed 2021-06-03 by github-actions[bot]