mcu-tools/mcuboot #541: Swap status documentation issue

solsbarry, 2019-08-30:
The documentation states that in the trailer the first 3 records are the swap status of the last sector, and the next 3 records are the status of last-1 sector. And so on. But I don't think that is true. The first 3 records are the status of the first `boot_swap_sectors` operation. But a single `boot_swap_sectors` operation can swap multiple sectors as long as the scratch partition is larger than 1 sector. `boot_copy_sector` should be named `boot_copy_sectors`, because it can copy multiple sectors. `boot_write_status` only writes status once following a call to `boot_copy_sector`.

so if the images you are swapping each 10 sectors large. and the scratch area is 2 sectors . Then 5 swap operations will take place (15 copy operations), and status will be written 3 times for each swap, or 15 total times. 
In the Swap status in the trailer,
The first 3 records will thus represent the swap status for sectors 8 and 9.
The next 3 records will represent swap status for sectors 7 and 6.
..
The last 3 records will represent swap status for sectors 1 and 0.

I am almost positive this is true. I am watching the memory get filled in in my debugger. Its possible I misunderstood the documentation. If that is the case then maybe it needs to be clarified.

Assuming I am correct,
I have yet to test whether this is an actual bug or just an error in documentation. I posted about a related issue that I think is actually a bug https://github.com/JuulLabs-OSS/mcuboot/issues/427 . has anyone actually tested that an interrupted swap can be recovered on a subsequent boot. I would hope so. That seems like core functionality of MCUboot. It is on my agenda for next week, as I slog through developing unit tests of my customized version of MCUboot.

utzig, 2019-09-09:
Yes, your assessment is mostly correct. It only has to save state for the swap operation which might be any size (and number of sectors) that the scratch area is able to fit. Also it only swaps sectors that effectively contain the image (based on the larger image among both slots). Regarding naming, I agree but I would rather go with `boot_copy_region` and `boot_erase_region` since the concept of sector(s) is not important for those functions.

> I am watching the memory get filled in in my debugger.

What do you mean by "memory get filled"?

> has anyone actually tested that an interrupted swap can be recovered on a subsequent boot

Of course, yes. Also check the code in the `sim` directory.

solsbarry, 2019-09-09:
I hadn't ever looked at the sim directory, thanks for pointing it out.

By "memory get filled", I meant I have the memory window in my IDE open, and as I step through the code, I watch the flash fill in in that window. So I see the status get written to the flash as the code executes. This is on Segger embedded studio, and using a NRF52840 chip.

PR #545 (2019-09-10): Use better naming for swap copy/erase functions
Since a copy/erase during a swap can result in some number of sectors erased and copied, which is not necessarily equal to 1, update design document and bootloader function naming to reflect this fact, by using the postfix "_region" instead of "_sector" in those cases.

This fixes #541
commit c28005b0 (2019-09-10): Use better naming for swap copy/erase functions

Since a copy/erase during a swap can result in some number of sectors
erased and copied, which is not necessarily equal to 1, update design
document and bootloader function naming to reflect this fact, by using
the postfix "_region" instead of "_sector" in those cases.

This fixes #541
closed 2019-09-10 by utzig