mcu-tools/mcuboot #550: Possibly a bug in boot_swap_sectors at boot/bootutil/src/loader.c: 1417

taovcu, 2019-09-11:
I see
rc = boot_write_status(state, bs); @ boot/bootutil/src/loader.c: 1417 
But I do not see the trailer sectors are erased before writing the status. 

A similar implementation has been verified as a bug in previous mcuboot veriosn, which is integrated in trusted-firmware-m 1.0beta.

utzig, 2019-09-11:
I assume you mean the trailers in the primary slot. They are erased either at: https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/bootutil/src/loader.c#L1318 or at https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/bootutil/src/loader.c#L1361. Which one happens depends on the size of the image (specifically if the last sector is used by the image).

taovcu, 2019-09-11:
@utzig  Thanks for the follow-up. 
The code that I tested is an old version loader.c, https://git.trustedfirmware.org/trusted-firmware-m.git/tree/bl2/ext/mcuboot/bootutil/src/loader.c?h=v1.0-beta 

The old version, after executing line 980, 'rc = boot_write_status(bs);', rc will be -1, my further debugging found the sector to write boot status was not erased, which caused the assert failure and  execution aborted.

I have not got a chance to test the current updated code. I do see some APIs changed. If you are pretty sure the problem I noticed in the old version was not actually a bug, or the bug has been fixed in new code, please close this issue. Thanks.

utzig, 2019-09-11:
Yes, I don't really follow forks, but if the bug existed I am pretty sure it would be caught by the simulator and probably would fail on HW as well. I suggest whoever is responsible for forks to at least stay in sync.

closed 2019-09-11 by utzig