mcu-tools/mcuboot #553: FW update problem

MarekPieta, 2019-09-13:
I use MCUBoot with 2 image slots on Zephyr device.
I noticed that image verification fails when overriding image with image with different size. After failed verification partition is erased, then image always works when reuploaded.
Is it necessary to erase whole partition when uploading new image? Or is it some sort of bug?

utzig, 2019-09-13:
> I noticed that image verification fails on reboot with image swap (test swap) after overriding image on second partition with image with different size. After failed verification, partition is erased.
On second upload try (after mentioned erase) image verification is successfull and image is swapped.

What version are you using? `master`? It would be very helpful if you could describe the exact steps you are doing.

> Is it necessary to erase whole partition when uploading new image? Or I observe some sort of bug?

No, you just need to erase whatever area you're writing the new image to, not the whole slot. When a verification of the image in second slot fails, it is erased, that is normal behaviour. So, it seems we have to find why it is failing verification.

MarekPieta, 2019-09-18:
I noticed that problem was caused by the fact that part of FLASH was not properly erased by application before write operation (it is why problem was not occurring after erasing whole partition). It is fixed already, so I am closing the issue. Sorry for problems.

closed 2019-09-18 by MarekPieta