mcu-tools/mcuboot #555: Does serial recovery mode need to abort if exited?

mike-scott, 2019-09-20:
I was experimenting with a delay during mcuboot which enters serial recovery mode for x seconds and then exits, but I notice that even if I remove the assert here:
https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/zephyr/main.c#L202

Things don't exactly "recover" themselves.  I tried cleaning up the serial recovery console a but, but that didn't help either.

Not sure what I'm missing.

carlescufi, 2019-09-23:
@nvlsianpu could you please take a look?

nvlsianpu, 2019-09-23:
@mike-scott As I understand your objective is to have serial recovery switched-on for a while first, and then boot the application regular. Am I right?

nvlsianpu, 2019-09-23:
> Does serial recovery mode need to abort if exited?

Serial recovery nowadays is designed to be exited only by reset.
I don't know about any thing which should be aborted.

As I know before exiting need to disable communication device used by the serial recovery.
(as a cure for zephyr power-management deficiencies): but this shouldn't be critical issue.

mike-scott, 2019-09-23:
> @mike-scott As I understand your objective is to have serial recovery switched-on for a while first, and then boot the application regular. Am I right?

Yes, that is correct.

closed 2020-11-06 by nvlsianpu