mcu-tools/mcuboot #576: Need to reset again after image swap for mcumgr to work correctly.

rickconrey, 2019-10-21:
Using STM32WB, mcuboot master, and zephyr master.

Uploading an image the first time works.  I can list, test, confirm, and reset.  Everything seems to work normally.  Once I have confirmed the uploaded image and reset the device to swap the image, the new image is loaded and runs fine.  If I try to upload a new image, I receive an UNKNOWN (1) error.

Work around:
Any time I want to upload new firmware, I must make sure the device has reset again after the last swap, then it seems to work.

Steps to make it work:
1. Upload image
2. confirm image and reset
3. image swaps and new firmware is running
4. reset again
5. now I can upload new firmware

nvlsianpu, 2019-10-23:
What is described above  must be zephyr mcumgr library or zephyr dfu subsystem issue. Mcuboot just chainnload the zephyr application in which issue was encountered so it doesn't control any application activities after that.

nvlsianpu, 2019-10-24:
I can't reproduce this with zephyr master on nRF52840_pca10056 board.

utzig, 2019-10-24:
@nvlsianpu What transport are you using?

nvlsianpu, 2019-10-24:
mcumgr over serial port - works out of the box with zephyr smp_svr  sample.

closed 2021-06-03 by github-actions[bot]