mcu-tools/mcuboot #581: Problem with flash_area_align() and BOOT_WRITE_SZ() mismatch for partitions in different flash devices.

solsbarry, 2019-10-29:
https://github.com/JuulLabs-OSS/mcuboot/blob/441e9978917724b993f7c98bf23466d63830ef5d/boot/bootutil/src/loader.c#L367

https://github.com/JuulLabs-OSS/mcuboot/blob/441e9978917724b993f7c98bf23466d63830ef5d/boot/bootutil/src/loader.c#L608

In` boot_read_status_bytes` it reads the trailer status entries in flash_area_read_is_empty assuming each status entry is` BOOT_WRITE_SZ(state)` wide.  The value of` BOOT_WRITE_SZ(state)` is set by `boot_write_sz()` and it is the larger of either the min write size of the scratch, or the min write size of the image primary partition.

If those two partitions are in the same flash device then that value is the same for them. But in my case where those partitions are in different flash devices with different minimum write sizes, I have a problem. The problem reveals itself when doing `boot_read_status_bytes`. On line 598 the status offset is calculated via  `boot_status_off` where the status entry size is found using the `flash_map` api via flash_area_align. However a few lines down on 608 the size of entries is assumed to be `boot_read_status_bytes`, instead of being found with the `flash_map` api as it was before.

You could fix this many ways, but I think the best would be to change line 608 to read 
`        rc = flash_area_read_is_empty(fap, off + i * flash_area_align(fap),`

if you do this, then I think all instances of ` BOOT_WRITE_SZ(state)` should be changed to `flash_area_align(fap)`.

Or you could do the reverse, and change all instances of `flash_area_align(fap)` to `BOOT_WRITE_SZ(state)`. except the problem with this is that there are many places that use `flash_area_align(fap)` that don't have access to state, such as `boot_status_off` in `bootutil_misc.c`

utzig, 2019-10-29:
Thanks, nice write up! AFAIR the reason for using scratch in internal flash is that it reveals the non-encrypted firmware; someone could start a swap and force interruptions while dumping the external flash and saving the blocks from scratch to recover the full non-encrypted image. If you are using encrypted images you should be aware of this fact. I think we can fix the issue but with a disclaimer about that it should not be used when the encrypted images option is enabled.

solsbarry, 2019-10-29:
Thanks.  I'm not using encrypted images, so it would be nice to have this work. But its good to know  about the security rational in case our feelings on encrypted images changed.

closed 2021-06-03 by github-actions[bot]