mcu-tools/mcuboot #586: NRF Cryptocell is not actually being disabled.

solsbarry, 2019-11-06:
https://github.com/JuulLabs-OSS/mcuboot/blob/8554ed5be658bf710b8e79cbac74e162c3b27338/ext/nrf/cc310_glue.h#L39

Is there a reason the NRF Cryptocell is not actually being disabled?

Currently
```
static inline void nrf_cc310_disable(void)
{
    NRF_CRYPTOCELL->ENABLE=1;
}
```
why not
```
static inline void nrf_cc310_disable(void)
{
    NRF_CRYPTOCELL->ENABLE=0;
}
```
**From the NRF52840 Manual V1.1**

[image]

utzig, 2019-11-06:
/cc @sigvartmh

sigvartmh, 2019-11-06:
Copypasta bug, will fix it. There is no actual reason it is not disabled. Thanks for spotting it :)

closed 2019-11-06 by utzig

solsbarry, 2019-11-06:
Thanks all, that was super fast.

PR #587 (2019-11-06): Bugfix: nRF Cryptocell not being disabled
Copy paste bug in cc310 glue layer for nRF where disable does not set
the enable register to 0. Thanks to @solsbarry for pointing this out.

Fixes: #586
commit 477ff5ba (2019-11-06): Bugfix: nRF Cryptocell not being disabled

Copy paste bug in cc310 glue layer for nRF where disable does not set
the enable register to 0. Thanks to @solsbarry for pointing this out.

Fixes: #586
commit 22f5432d (2019-11-19): Bugfix: nRF Cryptocell not being disabled

Copy paste bug in cc310 glue layer for nRF where disable does not set
the enable register to 0. Thanks to @solsbarry for pointing this out.

Fixes: #586