mcu-tools/mcuboot #601: zephyr: mcuboot is not being linked to the boot partition

SuperPrower, 2019-12-03:
0e3fa72df4849eb42c99efb34b8617046c2e33cb removed device tree overlay that is required to flash mcuboot to correct partition and adding a CMake variable that pointed to this file; however, without this, resulting image is not linked to the beginning of the flash:

```
...
(gdb) file ./mcuboot/boot/zephyr/build/zephyr/zephyr.elf
A program is being debugged already.
Are you sure you want to change the file? (y or n) y
Load new symbol table from "../mcuboot/boot/zephyr/build/zephyr/zephyr.elf"? (y or n) y
Reading symbols from ../mcuboot/boot/zephyr/build/zephyr/zephyr.elf...
(gdb) load
Loading section text, size 0xdc lma 0xc000 <- incorrect!
...
```

As far as I understand this is happens because ec134775d05e455d70f55339c06252115d62e638 made it such that Zephyr [will now link](https://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_USE_CODE_PARTITION.html)  application into `zephyr,code-partition`, which is  `&slot0_partition` for big portion of ARM boards. Thus, an overlay that will specify correct code partition is needed (which was deleted in commit mentioned at the beginning).

If my assumption is correct, I have a PR ready to fix this, but first I would like to hear opinion from more experienced people.

utzig, 2019-12-03:
/cc @mbolivar @nvlsianpu

d3zd3z, 2019-12-03:
0e3fa72 doesn't remove overlay support, it actually fixes it. You should still be able to create an overlay, at least as long as you are using a fairly recent version of Zephyr.

d3zd3z, 2019-12-03:
If you find that having an overlay named `boot/zephyr/boards/boarname.conf`, that will need to be fixed, though, although it might be a Zephyr issue.

SuperPrower, 2019-12-03:
> If you find that having an overlay named `boot/zephyr/boards/boarname.conf`, that will need to be fixed, though, although it might be a Zephyr issue.

Sorry, could you clarify this a bit? Do you mean that creating a board-specific overlay locally in location specified in your message is intended? If so, then instructions for Zephyr could use a clarification on that :)

d3zd3z, 2019-12-03:
The 0e3fa72 doesn't remove the overlay support that is present, it fixes it. With this change, the overlay files, already in the tree should now work. If you create a file `boot/zephyr/boards/boarname.overlay`, the Zephyr build system should use it. The path is based on where the application `CMakeLists.txt` file, which in this case is in `boot/zephyr`. The overlay support was added to Zephyr in a generic manner a while back, so we don't need to do anything special in MCUboot.

SuperPrower, 2019-12-03:
Yes, I understand how Zephyr looks for overlays currently and what this commit accomplished. I, however, was not talking about overlay _support_, instead, this issue is about lack of bundled overlay file itself. What I say is that currently most of the time (but not by default)  MCUboot will not be linked to the boot partition (0x00), instead, it will be linked to the slot0 partition since a lot of device configurations in Zephyr set it as their default code partition in the device tree:
```
chosen {
...
    		zephyr,code-partition = &slot0_partition;
}
```

And thus, due to how mcuboot is configured as Zephyr application, i.e. option USE_CODE_PARTITION, it uses this device tree configuration line and is linked to start at this partition, meaning it won't be flashed to the beginning of the flash.

One solution is, as you said, to have user define their `boot/zephyr/boards/boarname.overlay` with this option changed; if so, this probably should be mentioned in the README for Zephyr. Another solution, I believe, is to partially restore some of the code removed by mentioned commit, including overlay file with this option and CMakeLists.txt change to set `DTC_OVERLAY_FILE` variable that is looked for by Zephyr. 

I hope now it's clear what I'm trying to say. Thanks.

d3zd3z, 2019-12-03:
I'm not sure I understand. MCUboot should always be building in the boot partition. However, I'm finding that Zephyr builds seem to be broken, as it is failing to find `boot/zephyr/dts.overlay`. This was removed in error by 0e3fa72. I'll make a change to put that back, though.

SuperPrower, 2019-12-03:
@d3zd3z you understood perfectly what I was trying to (indirectly) convey in this message. I have a ready commit exactly for that :) I just wanted to see if I understood everything.

https://github.com/SuperPrower/mcuboot/commit/351e82382c86f9999b0b97617efc0b7392072029

I can make a PR if you want me to.

d3zd3z, 2019-12-03:
Can you go ahead and make a PR for your change? I tried just adding the overlay file back, and it appeared to work because of cmake's horrible caching behavior. I'll cancel my PR, and we can merge yours.

PR #604 (2019-12-05): zephyr: restore device tree overlay
0e3fa72df4849eb42c99efb34b8617046c2e33cb removed device tree overlay that configures Zephyr to link mcuboot at the beginning of the flash, and CMake variable that pointed to it (required for Zephyr). This resulted in mcuboot not being linked at some offset, rendering it unable to boot. Restore overlay and required code.

Without it, Zephyr will use `zephyr,code-partition` configuration from board device tree that usually points to the `slot0_partition`. This behaviour is due to the [CONFIG_USE_CODE_PARTITION](https://docs.zephyrproject.org/latest/reference/k [... cut]
closed 2019-12-05 by d3zd3z