mcu-tools/mcuboot #607: MCUboot doesn't actually support big endian

d3zd3z, 2019-12-04:
It is somewhat ambiguous what the intended endianness of various fields are within MCUboot. The `struct image_tlv` states in the comment that "All fields in little endian." The simulator explicitly calls out little endian when packing structures. It is likely we have only ever tested on a little endian platform.

This issue is here to track this. It is unclear when we might need big endian support, as little endian seems to overwhelming be used in embedded systems.

d3zd3z, 2019-12-11:
In addition, `imgtool.py` supports an `--endian` flag, but this clearly doesn't work. At least it doesn't match the comment about the little endian fields above `struct image_tlv`.

utzig, 2019-12-12:
/cc @schultetwin

schultetwin, 2019-12-15:
Sorry for the delay in response. I‘be moved away from the project using mcuboot in big endian mode. But I believe we’re still using it, and at least on the project that we’re using it on it’s working. I’ll ask more and get back to you.

d3zd3z, 2019-12-16:
@schultetwin Just a heads up, then, that a recent change d13318a14f99a5a3947bedc980020ec73800aaeb will probably cause problems for big endian targets. When implementing this, I followed the comments in the `image.h` where it says that the TLV header is always little endian. This is now the case with imgtool, but the length will change on big endian targets. We probably need to add a small amount of code to `tlv.h` to make this byte-swap the tag and the length values upon read.

schultetwin, 2019-12-22:
Ah. Yes, definitely. Thank you for the heads up. Agreed about the solution.

closed 2021-06-03 by github-actions[bot]