mcu-tools/mcuboot #616: Firmware update not working using OVERWRITE_ONLY option

rajkumarthiagarajan, 2019-12-10:
Hello,

In our project, we configured the scratch area size to 0, primary and secondary slots size are configured to same (0xF0000) and sector size for primary and secondary slots is 1 block (32768 bytes).

As part of firmware upgrade, we loaded the secondary slot with new firmware image. On reset when the BL2 executes, it always validates the primary slot and loads the image from the primary slot. It doesn’t do the image validation in the secondary slot. On debugging, I noticed that the BL2 calls a boot_slots_compatible() API. This API fails and throws the below warning 

**[WRN] Cannot upgrade: not all sectors fit inside scratch**
                
Due to this API failure, the swap type is set to BOOT_SWAP_TYPE_NONE. This results in skipping the validation of secondary slot and copying of image from secondary slot to primary slot 
if validation is successful. Instead it is always booting the image from primary slot all the time. 

Find below the code snippet from boot_slots_compatible() function implementation which displays the above warning message

if (sz0 == sz1) {
            primary_slot_sz += sz0;
            secondary_slot_sz += sz1;
            /* Scratch has to fit each swap operation to the size of the larger
             * sector among the primary slot and the secondary slot.
             */
            if (sz0 > scratch_sz || sz1 > scratch_sz) {
                BOOT_LOG_WRN("Cannot upgrade: not all sectors fit inside"
                             " scratch");
                return 0;
            }
            smaller = sz0 = sz1 = 0;
        }

In case of MCUBOOT_OVERWRITE_ONLY option, the scratch size will always be set to 0. I am not quite sure why there is a conditional check against scratch_sz which is expected to true  
always in case of MCUBOOT_OVERWRITE_ONLY option.

Regards,
T. Rajkumar

utzig, 2019-12-11:
This analysis is correct and it is already fixed on `master`. If you add the `#ifndef MCUBOOT_OVERWRITE_ONLY` checks from here: https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/bootutil/src/swap_scratch.c#L174, it won't be testing against `scratch_sz`.

d3zd3z, 2019-12-11:
Unless there is a need to backport this to an older release, I think we can close this. Please reopen if there is a need for a backport.

closed 2019-12-11 by d3zd3z