mcu-tools/mcuboot #631: Unable to use mcuboot with zephyr and nrf52840_pca10059 (dongle)

RIJU55, 2020-01-08:
Hi, I am trying the procedure for using **mcuboot with nrf52840_pca10059 (dongle)** as given in the documentation ( [https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10059/doc/index.html?highlight=nrf52840_pca10059]), I have achieved up to **step 4 of flashing the zip using** `nrfutil dfu usb-serial -pkg mcuboot.zip -p /dev/ttyACM0` but next when I do the subsequent steps **Boot into MCUboot serial recovery mode** by plugging the board in with the SW1 button pressed down I **do not get** the mentioned `/dev/serial/by-id/usb-ZEPHYR_MCUBOOT_0.01-if00 `as I am using Ubuntu 19.10 even my `/dev/ttyACM0` is **missing** and hence I am unable to perform the steps which ensue

utzig, 2020-01-08:
You do not have any file(s) under `/dev/serial/by-id/`? Also check that the board was detect with `dmesg`.

RIJU55, 2020-01-09:
no I do not have any file(s) under `/dev/serial/by-id/` will try` dmesg` again with mcuboot and get back to you, when I ran the zephyr/samples/subsys/usb/cdc_acm with nordic open bootloader I got the following on dmesg
```
[ 1749.112399] cdc_acm 3-3:1.0: ttyACM0: USB ACM device
[ 1758.442850] usb 3-3: USB disconnect, device number 6
[ 1758.443211] cdc_acm 3-3:1.0: failed to set dtr/rts
```

pjpetree, 2020-01-22:
I too am seeing the same thing.  I have used macOS, Linux and Windows hosts and no USB serial port ever shows up.

Any ideas?

pjpetree, 2020-01-23:
Found the issue.  It seems that there really should be a call to `usb_enable(NULL)` in main in order to properly initialize the USB serial port on the nrf52840_pca10059.

I have a fix, should I go ahead and submit a PR?

RIJU55, 2020-01-24:
Is that the solution if I just add `usb_enable(NULL)` it should run or do you have any further changes as well

pjpetree, 2020-01-24:
With just that change, the rest of the instructions work perfectly.

nvlsianpu, 2020-02-05:
fixed by #651

closed 2020-02-05 by nvlsianpu