mcu-tools/mcuboot #664: [Question] MCUBoot About Image Trailer

Visit-Learn, 2020-02-18:
Hi,

I am developer for MCU firmware and currently I am studying this MCUBoot mechanism and
I want to use this method into my FreeRTOS project and currently I think that I am on the right track
to study your code and flow, if possible I need group give me a hand to suggest me~

And my question is about image trailer, if I have a bin file image, how can I put this image
trailer into the end of image and I am also a little confused about Copy done and Image OK default
value when I need to flash image into device for first time by STLink~ Thank you so much.

[image]

I want to try my best to know this mechanism in order to porting this awesome mechanism into my
project and build tool~ Thank you again.

utzig, 2020-02-18:
You should be using `imgtool` which is under `scripts/imgtool.py` (|or install `pip install --upgrade imgtool`). It will add the required trailer entries when passing it `--pad`. The only fields you that need to be added are `MAGIC` and optionally `Image OK` (with value 0x01), the other fields are the bootloader's responsibility. This is not part of the image, but written to the end of the slot (for secondary slot when an upgrade is being asked for).

Visit-Learn, 2020-02-18:
Hi,

Thank you for reply. How could I add MAGIC and Image OK with value 0x01?
I check imgtool.py there has definition for IMAGE_MAGIC and boot_magic, do I need to modify
imgtool.py  IMAGE_MAGIC and boot_magic? And where I could add Image OK with value 0x01?

By the way, could I know the design concept for image why need to have two magic number? One is in header ih_magic and another is in trailer MAGIC.

Thank you~ And sorry to bother you~ : )

utzig, 2020-02-18:
> By the way, could I know the design concept for image why need to have two magic number? One is in header ih_magic and another is in trailer MAGIC.

This should be explained in docs/design.md, but one is to identify an image, the other is to identify an upgrade operation.

> to use imgtool.py and want to generate unsigned image first, the following is my testing command. But the error occur, the tool cannot detect --header-size and other options, it's like the imgtool.py version is different...

> python imgtool.py --header-size 0x80 --pad-header --align 8 --version 1.0.2 --slot-size 0x40000 testSrc.bin testDest.bin

`--header-size` was already present in the first `imgtool` versions. What's the output?

Visit-Learn, 2020-02-20:
Hi,

Sorry, the tool is work properly now, the command I typed it's incorrect.
Thank you so much.

And I saw you said I need to add optionally MAGIC and Image OK with value 0x01.
And I can find MAGIC setting in image.py with boot_magic variable, but I have no idea where
is Image OK settings, could you give me some advice? Thank you~ : )

henrikbrixandersen, 2020-02-20:
I have not found a way to set image_ok = 0x01 using imgtool either. Please advice.

utzig, 2020-02-21:
> I have not found a way to set image_ok = 0x01 using imgtool either. Please advice.

`imgtool` does not write image ok, only the magic field to start a test upgrade. I suppose you are a Zephyr user/developer from seeing you there, so the procedure would be to start a test upgrade and once your new image starts and runs, it can call `boot_write_img_confirmed` to confirm itself. But maybe we could do add a `--confirmed` flag to `imgtool` (to be used with `--pad`) to create a confirmed image? not sure why nobody asked for it...

utzig, 2020-02-21:
> And I saw you said I need to add optionally MAGIC and Image OK with value 0x01.
And I can find MAGIC setting in image.py with boot_magic variable, but I have no idea where
is Image OK settings, could you give me some advice? Thank you~ : )

I guess we could add a flag as I answered before, but to do it manually you would have to write the flash 8 bytes before MAGIC, so MAGIC being 16 bytes, you would write the value `0x01` at the index 24 before the end of the slot, as in, `<size of the slot> - 24 = 0x01`; sorry if that is not very clear, but a visual representation can be seen here: https://mcuboot.com/mcuboot/design.html#image-trailer

closed 2020-03-11 by utzig

azeemshatp, 2020-07-08:
> Hi,
>
> Sorry, the tool is work properly now, the command I typed it's incorrect.
> Thank you so much.
>
> And I saw you said I need to add optionally MAGIC and Image OK with value 0x01.
> And I can find [...]

Hi
Could you please suggest what was the correct command format here? I am getting into the same issue. "Error: no such option: --header-size"

Visit-Learn, 2020-07-08:
Hi,

imgtool.py **sign**

The keyword of **sign**  is needed, thank you~

azeemshatp, 2020-07-08:
> > > Hi,
> > > Sorry, the tool is work properly now, the command I typed it's incorrect.
> > > Thank you so much.
> > > And I saw you said I need to add optionally MAGIC and Image OK with value 0x01.
> > > And I can find MAGIC setting in image.py with boot_magic variable, but I have no idea where
> > > is Image OK settings, could you give me some advice? Thank you~ : )
> > 
> > 
> > Hi
> > Could you please suggest what was the correct command format here? I am getting into the same issue. "Error: no such option: --header-size"
> 
> Hi,
> 
> imgtool.py **sign**
> 
> The keyword of **sign** is needed, thank you~

Cool, I thought that's not required to create an unsigned image. Thanks

Visit-Learn, 2020-07-08:
> Cool, I thought that's not required to create an unsigned image. Thanks

Good~ : )

utzig, 2020-07-08:
> Cool, I thought that's not required to create an unsigned image. Thanks

`imgtool` did not support unsigned images originally so that's why the sub-command is called `sign`, but there is also an alias, `create`, so you can also run `imgtool create <bla bla bla>`.

azeemshatp, 2020-07-08:
> > Cool, I thought that's not required to create an unsigned image. Thanks
> 
> `imgtool` did not support unsigned images originally so that's why the sub-command is called `sign`, but there is also an alias, `create`, so you can also run `imgtool create <bla bla bla>`.

Cool. Is this documented somewhere?

utzig, 2020-07-08:
> Cool. Is this documented somewhere?

```
$ imgtool -h
Usage: imgtool [OPTIONS] COMMAND [ARGS]...

Options:
  -h, --help  Show this message and exit.

Commands:
  create   Create a signed or unsigned image INFILE and OUTFILE are parsed...
  getpriv  Dump private key from keypair
  getpub   Dump public key from keypair
  keygen   Generate pub/private keypair
  sign     Create a signed or unsigned image INFILE and OUTFILE are parsed...
  verify   Check that signed image can be verified by given key
  version  Print imgtool version information
```

PR #681 (2020-03-11): imgtool: add option for confirming a padded image
Add imgtool command line option for confirming (setting image_ok = 0x01) in a padded image.

Fixes: #664
commit 0ce958e7 (2020-03-11): imgtool: add option for confirming a padded image

Add imgtool command line option for confirming (setting image_ok =
0x01) in a padded image.

Fixes: #664