mcu-tools/mcuboot #668: chain-load the application via Reset

nvlsianpu, 2020-02-21:
objective: introduce  mechanism which allow to enter the application in the clean state as signaled in #529

possible solution:
Introduce execution flow in which mcuboot chain-load the application via reset.
Need to introduce a very early executed code partition which allows to recognize whether start to execute bootloader or the application.

utzig, 2020-02-21:
The challenge would be to avoid breaking test/revert flow. For other upgrade methods should be "easy" (or easier at least).

nvlsianpu, 2020-02-21:
^^ Decision what to execute should be done very early, before any image touch, so shouldn't affect even test/reset flow i believe. Just need to find some reliable way (software triggered resed prof) to recognize what should be executed.

d3zd3z, 2020-02-25:
Short of actual hardware support for this, the only thing I can think of is to put particular patterns in RAM. The trick is getting to only do this once (so I guess it should clear the pattern before then jumping to the app).

closed 2021-06-03 by github-actions[bot]