mcu-tools/mcuboot #669: Dependency check fails if if the image being depended on has BOOT_SWAP_TYPE_FAIL

solsbarry, 2020-02-21:
The issue is the the following line in
boot_verify_slot_dependency
https://github.com/JuulLabs-OSS/mcuboot/blob/310af87f61779f0f1930921055f2b19921a1ef45/boot/bootutil/src/loader.c#L1043-L1044

Situation:
There are 2 images.
Each has a version dependency on the other. 
There is a new version of each image in their respective secondary slots, and they depend on each other. The current version of each image will not satisfy the dependency of the new versions in the secondary slots. So both images need to upgrade or neither should upgrade.
The first image, secondary slot has a tlv that will not validate. So it gets assigned the swap type of BOOT_SWAP_TYPE_FAIL
The second image secondary is fine and gets assigned swap type BOOT_SWAP_TYPE_TEST

Problem
After the swap types are assigned the dependency check happens.
As you can see at the line referenced, the dependency check when determining which slot of the dependent image to look at, it only checks if the swap type for that image is BOOT_SWAP_TYPE_NONE, but it should also check if the type is BOOT_SWAP_TYPE_FAIL.

This is done at https://github.com/JuulLabs-OSS/mcuboot/blob/310af87f61779f0f1930921055f2b19921a1ef45/boot/bootutil/src/loader.c#L1153-L1155 when you figure out which slot of the current image to use. But it should also be done at the line mentioned above when you determine which slot to validate against.

so I would make the boot_verify_slot_dependency code look like:
```
dep_slot = (swap_type != BOOT_SWAP_TYPE_NONE &&
                swap_type != BOOT_SWAP_TYPE_FAIL) ?
                BOOT_SECONDARY_SLOT : BOOT_PRIMARY_SLOT;
```

I might be wrong, but I don't think so.

solsbarry, 2020-03-03:
@utzig any thoughts on this?

utzig, 2020-03-18:
> @utzig any thoughts on this?

Sorry for the late reply. Your analysis looks correct. I should also mention that `BOOT_SWAP_TYPE_PANIC` would incur in the same issue as `_FAIL` although that is somewhat unlikely to happen. Your suggestion of fix is OK but I would write it:

```
dep_slot = BOOT_IS_UPGRADE(swap_type) ? BOOT_SECONDARY_SLOT
                                      : BOOT_PRIMARY_SLOT;
```

I am trying to write a test case to the simulator to catch this. Thanks for pointing this out and feel free to submit a PR.

d3zd3z, 2020-03-30:
The fix appears to be merged, should we close this issue?

closed 2020-03-30 by solsbarry