mcu-tools/mcuboot #675: ecdsa verification fail

pakls, 2020-02-27:
imgtool.py can't verify signature if it is less than 70 or 71 bytes when ECDSA P256 is used.
It is caused by padding of signature to 72 bytes.

    def sign(self, payload):
        # To make fixed length, pad with one or two zeros.
        sig = self.raw_sign(payload)
        sig += b'\000' * (self.sig_len() - len(sig))
        return sig

But in image.py, the sign is called before signature TLV is added. Could we remove this padding?

utzig, 2020-02-27:
AFAIR the padding is required by the bootloader ecdsa validation code, would have to double check that.

solsbarry, 2020-02-27:
I think the signature should never be anything other than 70, 71 or 72 bytes for a ASN.1/DER formatted ECDSA 256 signature. If you have a lower length, you signature is likely in a different format. Maybe raw bytes with no encoding.

pakls, 2020-02-28:
Hi @solsbarry, yes, it ought to be one of 70,71,72.

The question here is: in the 'sign' method above, if the actual length of signature is 70, it pads two 0x00 behind it and declares this TLV as 72 bytes.

Then if we use 'verify' operation of 'imgtool.py', it would fail because imgtool.py take 72 bytes (according to TLV) for verification.

pakls, 2020-02-28:
The reason in git commit message is the 72 bytes must be declared in advance when signing is undertaken. I checked the imgtool.py and found it is not the case: signing doesn't include this TLV.

I tried to remove the padding and tried again: imgtool.py can verify properly.

I'm wondering whether the padding is necessary. Did I miss something somewhere?

pakls, 2020-03-03:
Hi @solsbarry, sorry, I just noticed my initial comment was wrong.

I actually wanted to describe that with the padding added by imgtool.py, if the signature length is 70 or 71 bytes, the verification would fail in imgtool.py because its 'verify' option only checks the len (which is 72 bytes) specified by TLV.

eg. if len(sig) is 70 => imgtool sign option pads 2 bytes, with tlv len is 72 => image created. Then imgtool verify option reads image => found tlv len is 72 => verify with len 72 => fails.

solsbarry, 2020-03-03:
That makes more sense. This isn't necessarily a problem with the image tool. In my mind its more of a problem with the verify function. The record is DER encoded, so that gives you the actual length of the signature. So the verify function should be able to ascertain the actual length of the signature and disregard the padding. But the verify might not be written that way.

pakls, 2020-03-04:
Hi @solsbarry, I tried removing the padding. Imgtool.py can correctly sign and verify the generated signature. I can't see the necessity in keeping this padding. Could I send a PR to remove it?

solsbarry, 2020-03-04:
I am nobody, so it doesn't matter what I think. I just use MCUboot, and mine is a heavily modified fork, so its hard for me to weigh in. But the most important thing is not whether imgtool.py can sign and verify signatures itself, but whether it can sign images and the firmware can verify. And the verify algorithm is not necessarily the same for all users. For me my verify function is fine with the imgtool signing. But hey put in a PR, if it makes sense to the people in charge it will go through, if not it wont.

d3zd3z, 2020-03-10:
So, the simple answer, is plainly, that padding is incorrect. It is not a valid ECDSA signature if it is padded, and we have done the padding in such a way that there is no way to remove it because it is just zero bytes, and these are permissible at the end of the signature. For example, mbed TLS will reject ECDSA signatures if the length is not correct.

I wrote up a plan to get rid of the padding, and it is sitting here in an unpushed tree. Let me see if I can get this together enough to fix this. The basic problem is that there are versions of MCUboot that enforce the padding being present. I added an option to imgtool.py initially to request the padding removed, and I was going to, in a future version, make no padding the default.

d3zd3z, 2020-03-10:
Back in December, I merged #621 which described the plan to remove the padding. At that time, the code was merged to eliminate the requirement that padding be present. Since we have had a release cycle of MCUboot now, I would propose at this time that we just make no padding the default in imgtool.py, and require an argument to add it back in. This will affect people running mcuboot 1.4.0 and earlier and would require this argument.

pakls, 2020-03-11:
> So, the simple answer, is plainly, that padding is incorrect.
agreed.

> It is not a valid ECDSA signature if it is padded, and we have done the padding in such a way that there is no way to remove it because it is just zero bytes, and these are permissible at the end of the signature.

It can be removed because the 2nd byte of the signature is length. This is what I did in my patch.

Try this website: https://lapo.it/asn1js/
Pasting this into the box: 304502206961, and decode it, I got message indicating the 2nd byte is length, instead of count:
*** Container at offset 2 has a length of 69, which is past the end of the stream ***

> For example, mbed TLS will reject ECDSA signatures if the length is not correct.

> I wrote up a plan to get rid of the padding, and it is sitting here in an unpushed tree. Let me see if I can get this together enough to fix this. The basic problem is that there are versions of [...]

Because embedded products tend to have longer life cycle. I think removing optional padding support would take quite some time. Meanwhile, my idea is to support trimming padding until complete eliminate could be done. How do you think?

d3zd3z, 2020-03-12:
I was incorrect about the length field in ASN.1. However, it is not just the second byte that indicates the length, the high bit of this byte indicates if a longer encoding is needed, which would be the case with longer values.

Regardless, this padding serves no purpose, and we need to just remove it.

pakls, 2020-03-13:
Thank for your response.

As you mentioned in a one comments above:

> and require an argument to add it back in.

I understand that the padding will be removed in the future. For some reason, an argument can be used to add it back in. One possible reason I could imagine is for backward compatibility for some earlier devices already have this padding in it. For example, their ROM code may recognizing this 72 bytes and can't be changed.

If the padding is added due to the reason above and one would like to verify the signature with imgtool.py, he/she can't do it because imgtool.py doesn't parse the ASN.1 length and, instead, passes 72 bytes to openssl. Then verification fails.

After being signed, the user won't have tool to verify the image, the only way is to download into a real device and let the device with ROM code supporting it to verify.

I just want to make the issue and its usage context clear during the discussion. If MCUBoot project wants to skip the need of verify a padded image. It is fine to me because I'll remove the padding anyway.

commit 04630cea (2020-03-13): fix #675 ECDSA verification fail when padding is added
closed 2020-03-13 by d3zd3z