mcu-tools/mcuboot #686: Getting mcuboot_swap_type as BOOT_SWAP_TYPE_REVERT during first reset after upgrade.

durgesh088, 2020-03-18:
I am using Zephyr on nrf52840 controller (nrf52840_pca10056 development kit).

I have written the sample application code which writes new image to slot 1, executes boot_request_upgrade(0) call and the reboots.

As per my understanding and according to mcuboot documentation, mcuboot_swap_type returns BOOT_SWAP_TYPE_TEST and user need to confirm the image. But I am getting swap type as BOOT_SWAP_TYPE_REVERT after upgrade. I am getting BOOT_SWAP_TYPE_REVERT during first boot after upgrade.

Below are the corresponding logs.

> *** Booting Zephyr OS build zephyr-v2.2.0-453-g0278f9bc72b0  ***
> [00:00:00.000,122] <inf> DFU: MCUBOOT_SWAP_TYPE : 1
> 
> [00:00:00.000,152] <inf> DFU: boot_img_hdr 
> [00:00:00.000,152] <inf> DFU: 1.
> [00:00:00.000,152] <inf> DFU: 4.
> [00:00:00.000,152] <inf> DFU: 0.
> [00:00:00.000,152] <inf> DFU: 0
> [00:00:00.000,152] <inf> DFU: 
> 
> [00:00:00.000,183] <inf> DFU: src_img_hdr 
> [00:00:00.000,183] <inf> DFU: 1.
> [00:00:00.000,183] <inf> DFU: 4.
> [00:00:00.000,183] <inf> DFU: 1.
> [00:00:00.000,183] <inf> DFU: 0
> [00:00:00.000,183] <inf> DFU: 
> 
> [00:00:01.323,852] <inf> DFU: NRF_FLASH_DRV_NAME size 65536 off 180224
> 
> [00:00:02.116,882] <inf> DFU: Copied bytes 65536, status : 0
> 
> [00:00:02.117,126] <inf> DFU: COPYING OF IMAGE IS DONE. WAIT FOR 10 SECONDS AND REBOOT
> 
> [00:00:02.117,126] <inf> DFU: REBOOTING
> 
> *** Booting Zephyr OS build zephyr-v2.2.0-453-g0278f9bc72b0  ***
> [00:00:00.000,152] <inf> DFU: MCUBOOT_SWAP_TYPE : 4
> 
> [00:00:00.000,152] <inf> DFU: BOOT SWAP TYPE IS REVERT
> 
> [00:00:00.000,152] <inf> DFU: boot_img_hdr 
> [00:00:00.000,152] <inf> DFU: 1.
> [00:00:00.000,152] <inf> DFU: 4.
> [00:00:00.000,183] <inf> DFU: 1.
> [00:00:00.000,183] <inf> DFU: 0
> [00:00:00.000,183] <inf> DFU: 

I have generated the images using below commands

> /home/durgesh/mcuboot_test/mcuboot_src/scripts/imgtool.py sign --key /home/durgesh/mcuboot_test/mcuboot_src/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.4.0 --slot-size 65536 --pad /home/durgesh/MCUBOOT/MCUBOOT.bin /home/durgesh/MCUBOOT/MCUBOOT.signed.1.bin
> /home/durgesh/mcuboot_test/mcuboot_src/scripts/imgtool.py sign --key /home/durgesh/mcuboot_test/mcuboot_src/root-rsa-2048.pem --header-size 0x200 --align 8 --version 1.4.1 --slot-size 65536 --pad /home/durgesh/MCUBOOT/MCUBOOT.bin /home/durgesh/MCUBOOT/MCUBOOT.signed.2.bin

And flashed the code with

> loadfile mcu-boot.bin 0x0
> loadfile MCUBOOT.signed.1.bin 0xC000
> loadfile MCUBOOT.signed.2.bin 0x2C000

I have attached the zephyr application used for upgrade.

[zephyr_dfu_app.zip](https://github.com/JuulLabs-OSS/mcuboot/files/4349452/zephyr_dfu_app.zip)

utzig, 2020-03-18:
> I am getting BOOT_SWAP_TYPE_REVERT during first boot after upgrade.

Do you really mean this? After upgrade you are in test mode, so a revert is expected. Also add the bootloader debug logs.

durgesh088, 2020-03-18:
Can you please clarify my below question.

After successful upgrade, during first boot if I execute mcuboot_swap_type() in my application what should it return ?
BOOT_SWAP_TYPE_TEST or BOOT_SWAP_TYPE_REVERT.

utzig, 2020-03-18:
> Can you please clarify my below question.
> 
> After successful upgrade, after first boot if I execute mcuboot_swap_type() in my application what should it return ?
> BOOT_SWAP_TYPE_TEST or BOOT_SWAP_TYPE_REVERT.

`BOOT_SWAP_TYPE_REVERT`

durgesh088, 2020-03-18:
I have understood the documentation like this.

After successful upgrade, `mcuboot_swap_type()` executed from application returns `BOOT_SWAP_TYPE_TEST`. If user doesn't confirm the image by executing `boot_write_img_confirmed()`, then mcuboot swaps the images of slot 0 and slot 1 during subsequent boot and boots with swap type as `BOOT_SWAP_TYPE_REVERT`

But as per your comment, application gets swap type as `BOOT_SWAP_TYPE_REVERT` after upgrade and not `BOOT_SWAP_TYPE_TEST`.

Is there any sample application code which gives better understanding of this ?

utzig, 2020-03-18:
Maybe we don't agree what "after successful upgrade" means, `BOOT_SWAP_TYPE_TEST` is before an upgrade when an image is marked as upgrade. After the upgrade the **test** was already done and it switches to **revert**. Honestly I have no idea why you're even checking this in your app, since it's a bootloader responsibility, but I suppose you must have your reasons.

durgesh088, 2020-03-18:
I am checking swap type to confirm the upgraded image by executing `boot_write_img_confirmed()`.
Do you suggest to confirm the upgraded image (through application) irrespective of swap type ? As you have mentioned that swap type is bootloader's responsibility.

utzig, 2020-03-18:
> I am checking swap type to confirm the upgraded image by executing `boot_write_img_confirmed()`.
> Do you suggest to confirm the upgraded image (through application) irrespective of swap type ?, As you have mentioned that swap type is bootloader's responsibility.

You can just call `boot_write_img_confirmed` like you're doing, there is no need to do any checking.

closed 2020-04-17 by durgesh088