mcu-tools/mcuboot #689: Unable to compile MCUBoot for Zephyr 2.2.0 with DFU enabled

bogdanm, 2020-03-19:
While trying to compile the latest revision of MCUBoot (4878c27c82d03b36c46c34ef45e0173730ff72ad) with Zephyr release 2.2.0 (https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v2.2.0) the linker fails because some functions are defined in both projects, for example:

* `flash_area_erased_val`
    - https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/zephyr/flash_map_extended.c#L109
    - https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.2.0/subsys/dfu/boot/mcuboot.c#L209
* `flash_area_read_is_empty`
    - https://github.com/JuulLabs-OSS/mcuboot/blob/master/boot/zephyr/flash_map_extended.c#L115
    - https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.2.0/subsys/dfu/boot/mcuboot.c#L218

(other such functions exist)

nvlsianpu, 2020-03-19:
zephyr DFU modules should been disabled wile you compile mcuboot project. Will check tomorrow why they are enabled.

nvlsianpu, 2020-03-20:
Reproduced.
Reason is that USB DFU class uses zephyr code which re-implement these two mcuboot functions.

Solution is to extend flas_area API in the zephyr code base by these function which are already in the zephyr code-base but reserved only for `dfu/boot/mcuboot` submodule:
* `uint8_t flash_area_erased_val(const struct flash_area *fa);`
* `int flash_area_read_is_empty(const struct flash_area *fa, uint32_t off,
	void *dst, uint32_t len);`

@mbolivar Will you accept such extension?

pabigot, 2020-03-23:
There's some reference to this in zephyrproject-rtos/zephyr#20744, it would be nice if somebody fixed this properly.

closed 2021-06-03 by github-actions[bot]