mcu-tools/mcuboot #697: zephyr nrf52840 mcuboot compile issue

Narendhar95559, 2020-03-27:
Hi,

Iam using setup zephyr(master-v2.2.99) with sdk(0.11.2) and mcuboot(master)
when trying to compile zephyr boot getting below error:-
In file included from ../main.c:29:
/home/dell/work/mcuboot/boot/bootutil/include/bootutil/image.h:137:1: warning: data definition has no type or storage class
  137 | BUILD_ASSERT(sizeof(struct image_header) == IMAGE_HEADER_SIZE,
      | ^~~~~~~~~~~~
/home/dell/work/mcuboot/boot/bootutil/include/bootutil/image.h:137:1: error: type defaults to 'int' in declaration of 'BUILD_ASSERT' [-Werror=implicit-int]
cc1: some warnings being treated as errors
[48/232] Building C object zephyr/CMakeFiles/zephyr.dir/home/dell/work/zephyrproject/modules/debug/segger/rtt/SEGGER_RTT.c.obj
ninja: build stopped: subcommand failed.

nvlsianpu, 2020-03-27:
Awaiting on https://github.com/zephyrproject-rtos/zephyr/pull/23625

dnsglk, 2020-03-28:
I wonder why this change was made in MCUBoot before the corresponding commit made it to the zephyr...

nvlsianpu, 2020-03-28:
@dnsglk Commit was merged too early.

Narendhar95559, 2020-03-30:
@nvlsianpu can you suggest commit id to checkout for mcuboot to work on zephyr(2.2.99)

d3zd3z, 2020-03-30:
It shouldn't be difficult to revert the problematic patch. Do we need to push this through, as it appears it still hasn't merged into Zephyr?

dnsglk, 2020-03-30:
@Narendhar95559 reverting back to this commit 274c5da95 worked for me just fine!

nvlsianpu, 2020-03-31:
@Narendhar95559 Fixed, now.

closed 2020-03-31 by nvlsianpu