mcu-tools/mcuboot #706: Bootstrap does not work when CONFIG_BOOT_SWAP_USING_MOVE=y

DunderRoffe, 2020-04-02:
I am trying to bootstrap MCUBoot using Slot1 but it does not seem to work when CONFIG_BOOT_SWAP_USING_MOVE=y. 

It seems like it fails on checking the magic for Slot 0. This is expected as there is nothing in Slot0 but I do not understand why it is checked.

Is this a bug in MCUBoot or have I misunderstood how this is supposed to work?
It works fine when CONFIG_BOOT_SWAP_USING_MOVE=n

RTOS: Zephyr
Target board: particle_boron
mcuboot conf:
```
CONFIG_BOOT_BOOTSTRAP=y
CONFIG_BOOT_SWAP_USING_MOVE=y
```

erik-johnson-nimbelink, 2020-10-02:
I believe I've also come across this. I haven't been able to get bootstrapping to work with this combination.

closed 2020-10-05 by utzig

utzig, 2020-10-05:
Thanks for reporting, it should have been fixed by #826, feel free to reopen if the issue persists!

PR #826 (2020-10-05): sim: add bootstrap test
This should fix bootstrap in swap-move along with other fixes.

@d3zd3z After some issues I've found working on this PR, I think we should just remove the `MCUBOOT_OVERWRITE_ONLY_FAST` option and make it the default behavior when doing overwrite-only, what do you think?

Fixes #706, #738