mcu-tools/mcuboot #709: SAM E70 has a write size > BOOT_MAX_ALIGN (16 bytes)

utzig, 2020-04-02:
See: http://ww1.microchip.com/downloads/en/DeviceDoc/SAM-E70-S70-V70-V71-Family-Data-Sheet-DS60001527D.pdf . As per page 135:

```
22.4.3.2.4 Programming Bytes
Individual bytes can be programmed using the Partial Page Programming mode.
In this case, an area of 128 bits must be reserved for each byte.
Refer to the Figure 22-11
```

This is going to cause swap upgrades to not work due to current `BOOT_MAX_ALIGN` equal to 8. Possibly also an issue with other SAMx models.

d3zd3z, 2020-04-10:
As far as I can tell, this device has 16-byte write alignment, and also 128K erase sectors. Given the large sector size, it might be possible to make the max align larger for it, but the swap state will end up fairly large (although the number of sectors will be smaller).

I think I'll create another issue to capture flash characteristics of devices we currently can't support with MCUboot: #713

closed 2021-06-03 by github-actions[bot]

utzig, 2022-11-10:
> Hi everyone i have same problem when i try to do update over canopen. Did someone fix this?

It should be working now if you are using the tip of `main`, at least with Zephyr.