mcu-tools/mcuboot #726: Question: What is needed to add SPI flash with "Little FS" file system?

Aukstkalnis, 2020-04-24:
Hello. We are using zephyr project for our new project's development. We want to make MCUbool be able to communicate with external flash and be able to use Little FS file system that is in that external flash chip. I need some guidance on where to start and what is needs to be changed.

Aukstkalnis, 2020-05-04:
Any suggestions? If you can't suggest anything it would be nice to know :)

utzig, 2020-05-04:
> Any suggestions? If you can't suggest anything it would be nice to know :)

MCUBoot is a bootloader, it checks if an image is valid and boots into it if it is. It does not do any FS related operations, like opening, reading or writing files/dirs. Your question would better be directed to the OS you're using (for Zephyr: https://github.com/zephyrproject-rtos/zephyr).

Aukstkalnis, 2020-05-04:
I understand that. The point is that i want MCU Boot to use external flash as second partition and not the internal. In external flash there is file system. What i want to do is able download fw with zephyr app, store it to external flash and trigger restart. MCU Boot scans for file in external flash and does the same what it does now - validate new image and swaps old image with new. I would like to get some tips on what to be aware of when developing this feature on MCU boot fork.

utzig, 2020-05-04:
As long as the external flash is supported and you adjusted the DTS properly there's not much else, MCUBoot won't touch anything else. Maybe this might be helpful: https://github.com/JuulLabs-OSS/mcuboot/wiki/Using-the-nrf52840_pca10056-SPI-flash-with-Zephyr

Aukstkalnis, 2021-05-20:
It was actually quite simple to use second and scratch partitions in external SPI flash driver. Thanks for pointing that link @utzig. I jus needed to enable some config parameters and fix my board's dts flash and MCU Boot was able to use SPI flash :) So thank you 👍
Sorry for the late response. I forgot about this issue. If you do not intend to keep this issue open for reference then you can close it.