mcu-tools/mcuboot #756: Cannot build for Zephyr 2.2.0

kubiaj, 2020-06-16:
Hi,
the current master (33fbef51c02b169607949544502b3907fb708f37)  version is not possible to compile with Zephyr 2.2 as it reports following error 

```
mcuboot/boot/zephyr/build $ cmake -GNinja -DBOARD=asd ..
CMake Error at CMakeLists.txt:44 (find_package):
  Could not find a package configuration file provided by "Zephyr" with any
  of the following names:

    ZephyrConfig.cmake
    zephyr-config.cmake

  Add the installation prefix of "Zephyr" to CMAKE_PREFIX_PATH or set
  "Zephyr_DIR" to a directory containing one of the above files.  If "Zephyr"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!

```
This is caused by following commit 

43004b85419b370154b681d3968f38f4c17d6d1b  - cmake: use find_package to locate Zephyr  

By simply reverting this commit is possible to compile the code for Zephyr 2.2.  Would it be possible to add some run conditions to maintain Zephyr 2.2 backward compatibility?

kubiaj, 2020-06-16:
Sorry I was wrong, there are more compilation issues against Zephyr 2.2  even with latest release 1.6. So the question is if it will be ever supported? I don't even mention the Zephyr 1.14.x which is LTS.

nvlsianpu, 2020-06-17:
https://github.com/JuulLabs-OSS/mcuboot/blob/master/docs/release-notes.md#zephyr-rtos-compatibility

kubiaj, 2020-06-18:
Hi, 
nice but there no such branch already. The announced commit 1a89ca1238 while the 1.6 was released was even before the 2.3 final release, so you have give up Zephyr 2.2 user even before was the 2.3 release. I was hoping that mcuboot will be more or less independent on the fury Zephyr develop. Lets say that the most of the code in mcuboot is Zephyr independent. 

  Unfortunately we (and I do believe that many companies) don't have the luxury to switch to a newer version of Zephyr once it is released and we need a stable piece of code - switch to another version always mean some pain (we have moved from 1.14 to 2.2 and it took us a long time). 

Would be better to thing abut the backward compatibility during the code update. For example the commit  43004b8  - it is a one line of code and it totally break the compatibility. Adding simple if could solve it pretty much quickly. I do know that this will need a human resources, but maybe some guidelines would help here. 

I don't believe that 2.2 vs 2.3 have so many differences that it will need a new branch which will not be maintained anyway.

I'm willing to help here to address backward compatibility issues, but my knowledge of Cmake is highly limited and I dont know which hard rock I should use for Zephyr version determination.

nvlsianpu, 2020-06-18:
Do you know that zephyr is using  exact mcuboot reference? https://github.com/zephyrproject-rtos/zephyr/blob/zephyr-v2.2.0/west.yml#L80

Policy in mcuboot master is to be compatible with zephyr master at the time. Release is being done from the master.

nvlsianpu, 2020-06-18:
https://github.com/nvlsianpu/mcuboot/tree/revert-zephyr-2.2.0-compatible this was my try to create bacward compatible branch at the time of last release.

nvlsianpu, 2020-06-18:
For 1.6.0 need to revert #b0890aeaa4433bb36bbb9c564736cfc88ffcfd33 #45b00acf0fa5e89880407b639578066cefcd41fa #00108fe9730a2b06de4c7141afbe6155ed51c83c 7024340b6627ce19998d1beb19fc211d7d0e6a68 #6e26e38585cc28ee2b22077740507546fe909dc9 and the same what was reverted in ^^ pointed branch.

kubiaj, 2020-06-18:
Many thanks, I will try it.

Another question related to this: we are willing to add some features add/update as we need some additional features IF we receive a green light from 3rd party code suppliers. What will be to workflow as we have stick to the 2.2 version. I can port it to the master & 2.xx Zephyr, but we will still need the 2.2 compatibility which will be now done by hacking current master/release.

nvlsianpu, 2020-06-18:
for the current master master (33fbef5) revert in addition:
#33fbef51c02b169607949544502b3907fb708f37
#0813efeac65588cc3895ba71cdc69c88bb946334
#43004b85419b370154b681d3968f38f4c17d6d1b

nvlsianpu, 2020-06-18:
I'm not administrator here (anyway I don't see problem with having backward compatibility branch) @d3zd3z?

utzig, 2020-06-18:
> I'm not administrator here (anyway I don't see problem with having backward compatibility branch) @d3zd3z?

You have write permissions, so you should be able to push a branch if you want.

nvlsianpu, 2020-06-18:
@kubiaj does the revert commit list works for you?

nvlsianpu, 2020-06-18:
I did branch https://github.com/JuulLabs-OSS/mcuboot/tree/v1.6.0-based-zephyr-2.2.0-compatible
This is mcuboot compatible with zephyr 2.2.0 version based on mcuboot 1.6.0

kubiaj, 2020-06-18:
@nvlsianpu I have to run away now. I did quick test but I run into some conflict so I need to figure a way how to get rid of them. I will check it lately again. 

I halve also checked your branch - I had to tweak it a bit and I have successfully compile it.  But just now I realized that I didnt switch to the revert-zephyr-2.2.0-compatible branch, argh ... So I will test it again....

kubiaj, 2020-06-18:
@nvlsianpu many thanks the mcuboot v1.6.0-based-zephyr-2.2.0-compatible seems to be working (I can compile it - but not tested with HW)  I have also ported there the single image feature which I needed.

closed 2021-06-03 by github-actions[bot]