mcu-tools/mcuboot #761: Creating signed image with imgtool

michaelthomasj, 2020-06-30:
Hi, 
I am trying to use imgtool to create a signed image and am running into a couple of issues. 
Attached is the address space for the particular MCU in question
[image].
As you can see, the flash area is at the very bottom and the Cortex-M System registers are at the very top. 
The imgtool only takes in a bin or hex input. 
When I output a .bin from my toolchain, the image is very big since there is some data being put into the System for Cortex-M area as well as the bottom flash with everything in between as 0. So, using this .bin as is is not possible. 
I then got my toolchain to output an Intel Hex file, but once that file is signed by imgtool, the format is not recognized by the debugger (Segger JLink) anymore. 
I looked at the header that was added and it doesnt quite look like it follows the Intel Hex specification ?
Attached are the original and signed hex files. 
[image_creation.zip](https://github.com/JuulLabs-OSS/mcuboot/files/4849351/image_creation.zip)

What is the best avenue to progress?
Thanks
MIchael T

nvlsianpu, 2020-06-30:
I see - your output looks erroneously - imgtool processed your input file badly.
But, indead the intel-hex file you attempted to sign as single image should contain continuous binary vector.
Your hex file seams to contain a few (depended each other) binaries.
If you have multiple image (Are these areas really possible update part?) you need to sign each of them fallowing mcuboot multi-mage feature.

michaelthomasj, 2020-07-02:
Thank you for the response. I was able to work around it. 
Regards

closed 2020-07-02 by michaelthomasj