mcu-tools/mcuboot #762: DFU failure due to unexpected/invalid trailer in secondary slot

nwsetec, 2020-07-03:
I'm getting an occassional issue where a DFU transfer to the secondary slot (when the primary slot application is running) can't proceed because even though the secondary slot is otherwise empty (all 0xFF) there is a trailer at the end.  The trailer looks like this:

```
0x00065FE8: FFFFFF01
0x00065FF0: F395C277
0x00065FF4: 7FEFD260
0x00065FF8: 0F505235
0x00065FFC: 8079B62C
```

Bytes FFFFFF01 indicates the image is 'ok' but it is otherwise empty.

I'm using a nRF52832 SoC with an external flash that is used for the secondary slot.
The bootloader was built with mcuboot tag v1.3.1 and
Zephyr v1.14.1 (9591025942af9d8334901cc281d2b41a1747836d)
I had to revert mcuboot commit f99a4c790a6b4f508938f36e811c80ae1f25c2a3 to get it to build.
We also had to modify mcuboot to ensure an external WDT was regularly cleared.

This is a showstopper for us as if this failure occurs in the field the device can not longer receive updates.

I'm looking into implementing a workaround by modifying the mcumgr code in Zephyr 1.14.1 to add an extra check at the begining of a DFU transfer that the image trailer is valid by also checking if the 1st sector of the secondary slot is empty and if it is then we'll erase the last sector of the secondary slot and let the DFU transfer commence.

Note also that mcuboot doesn't seem to detect the invalid trailer and erase it on reboot.

I haven't been able to find any other issues that look like this issue in the mcuboot project.

utzig, 2020-07-03:
What external flash are you using?

nwsetec, 2020-07-03:
@utzig a Spansion S25FL216K

utzig, 2020-07-04:
> Note also that mcuboot doesn't seem to detect the invalid trailer and erase it on reboot.

It does detect it but does not erase it, maybe we should erase it but it isn't required. I can't see no possible way for this to happen, unless there is some issue in the erase operation of the flash driver (which is why I asked about the driver). The code which erases the trailer in the secondary slot is well protected against interruptions and it happens always when starting a new upgrade. Anyway I just tried to write a valid magic + image_ok to an erased secondary here and it is properly handled. Although it is initially detected as a swap upgrade request, later the secondary is run through a validation routine which fails (because it is erased) and this ends up in just booting the primary slot. I run it on mcuboot `master` which is quite different but it didn't change in essence for those steps I mentioned (and I checked v1.3.1's source code as well).

Some extra information could be helpful:

1) Is there any trailer in the primary slot?
2) Could you post your bootloader config?
3) If you have a "bricked" device, please post a backtrace up to the brick operation.

nixward, 2020-07-04:
Thanks @utzig I'll try to get what I can for you ASAP

nwsetec, 2020-07-05:
The trailer in the primary slot looks like this:
```
0x00065FE8: FFFFFF01
0x00065FF0: FFFFFFFF
0x00065FF4: FFFFFFFF
0x00065FF8: FFFFFFFF
0x00065FFC: FFFFFFFF
```

[mcuboot_conf.txt](https://github.com/JuulLabs-OSS/mcuboot/files/4874840/mcuboot_conf.txt)

nwsetec, 2020-07-06:
@utzig 
I just discovered that the trailer I decribed in the issue description is missing an 'erased' word (`0x00065FEC: FFFFFFFF`) so is misleading.  It should be:

```
0x00065FE8: FFFFFF01
0x00065FEC: FFFFFFFF
0x00065FF0: F395C277
0x00065FF4: 7FEFD260
0x00065FF8: 0F505235
0x00065FFC: 8079B62C
```

nwsetec, 2020-07-06:
@utzig a theory I have but I haven't figured out a reliable way to test this.

In the system we have there are up to four devices connected over Bluetooth LE and any one of them can start a DFU.  I've read through the Zephyr v1.14.1 mcumgr code and there doesn't appear to be any lock provided to the DFU initiator over the DFU service.  The hypothetical failure senario that I'm thinking of is that one device is uploading to the secondary slot, finishes and actions 'Confirm only' (which I believe causes the trailer to be written to slot-1) but sneaking inbetween the upload and the trailer write operations a second device starts a DFU (with a different DFU image) that erases slot-1 and then fails leaving the secondary slot erased but not the trailer as described in this issue.

Can you imagine the above hypothetical failure senario being possible?

utzig, 2020-07-06:
> Can you imagine the above hypothetical failure senario being possible?

Yes, but it would not explain the bricking, which is what I am most concerned with.

DeclanTraill, 2020-07-07:
@nwsetec Sounds like a plausible explanations except that if the 2nd slot was erased by the second device, wouldn't that erase the written Trailer from the first device, thus preventing the lockup condition?

utzig, 2020-07-07:
> [mcuboot_conf.txt](https://github.com/JuulLabs-OSS/mcuboot/files/4874840/mcuboot_conf.txt)

Why are you using `CONFIG_MULTITHREADING=y`?

nwsetec, 2020-07-07:
> Why are you using `CONFIG_MULTITHREADING=y`?

I actually don't know why that was enabled.  In theory it shouldn't cause a problem though right?  The image size would just be a bit bigger.

utzig, 2020-07-07:
> In theory it shouldn't cause a problem though right?

No idea. I know there are some targets that enable it because the flash driver does not work without it, eg, STM32, but I never seen a similar issue in those either. I was just curious...

nwsetec, 2020-07-09:
Thanks for your help with this @utzig!

closed 2021-06-03 by github-actions[bot]