mcu-tools/mcuboot #800: Single-image boot is ambiguous

davidvincze, 2020-09-14:
The terms "single-image boot" and the "single image application" of Zephyr are confused.
The original suggestion comes from @nvlsianpu (https://github.com/JuulLabs-OSS/mcuboot/pull/793#discussion_r487111576).
It is not clear whether the MCUBoot handles only one updateable image with the "classic" primary + secondary image slot model or there is only 1 image slot in total.
Would it make sense to have this single slot/image mode in case of multiple independent images?
The documentation and comments should be rephrased to clearly distinguish these 2 cases.

davidvincze, 2020-09-14:
I would rather rename the `MCUBOOT_SINGLE_IMAGE_DFU` option to something like `MCUBOOT_SINGLE_SLOT_DFU`.

nvlsianpu, 2020-09-15:
@de-nordic What is your opinion? As you are the inventor of single slot DFU.

closed 2020-09-16 by nvlsianpu

davidvincze, 2020-09-16:
It was fixed by: https://github.com/JuulLabs-OSS/mcuboot/pull/801.
I am closing this issue.

PR #801 (2020-09-16): zephyr: renamed single-image mode to single-slot mode
The terms "single-image boot" and the "single image application"
of Zephyr are confused. It might be might be understood as one-pair image
dual-bank mode in comparison to multi-image mode.

This patch replaces CONFIG_SINGLE_IMAGE_DFU by CONFIG_SINGLE_SLOT_DFU
name, which makes this mode name unambiguous.

fixes #800

zephyr CI: https://github.com/zephyrproject-rtos/zephyr/pull/28396
commit fdff3e1e (2020-09-16): boot/zephyr: renamed single-image mode to single-application-slot mode

The terms "single-image boot" and the "single image application"
of Zephyr are confused. It might be might be understood as one-pair image
dual-bank mode in comparison to multi-image mode.

This patch replaces CONFIG_SINGLE_IMAGE_DFU by
CONFIG_SINGLE_APPLICATION_SLOT name,
which makes this mode name unambiguous.

fixes #800

fix