mcu-tools/mcuboot #829: Brick when using BOOT_SWAP_USING_MOVE and reset happens during images swap

mniestroj, 2020-10-05:
**Describe the bug**
Interrupted swap of images when **swap_move** strategy is selected, can end up in "bricked" state. Possible workaround is to use **swap_scratch** strategy, which doesn't seem to have this specific issue.

**To Reproduce**
See https://github.com/zephyrproject-rtos/zephyr/issues/28933 for reproducing using Zephyr RTOS.

**Expected behavior**
Swap of images should continue.

**Impact**
Remote firmware upgrade is not safe, i.e. can result in bricked devices in the field.

**Environment (please complete the following information):**
 - MCU software: Zephyr RTOS
 - OS: Linux
 - Toolchain: Zephyr SDK 0.11.4
 - aac99a14c9cdc8684fd4fbced130d3065a75ffea (tip of master) + https://github.com/JuulLabs-OSS/mcuboot/pull/828

**Additional context**
It seems that problems start here (https://github.com/JuulLabs-OSS/mcuboot/blob/aac99a14c9cdc8684fd4fbced130d3065a75ffea/boot/bootutil/src/loader.c#L1084):
```
        /*
         * If a swap was under way, the swap_size should already be present
         * in the trailer...
         */
        rc = boot_read_swap_size(image_index, &bs->swap_size);
        assert(rc == 0);
```
mcuboot reads `0xFFFFFFFF` after swap was interrupted, because this is what was contained originally in `app.hex` (signed and confirmed) image trailer and that was never updated by mcuboot. This value is passed as `copy_size` to `swap_run()` and results in infinite loop there (https://github.com/JuulLabs-OSS/mcuboot/blob/aac99a14c9cdc8684fd4fbced130d3065a75ffea/boot/bootutil/src/swap_move.c#L448):
```
    while (1) {
        sz += sector_sz;
        /* Skip to next sector because all sectors will be moved up. */
        g_last_idx++;
        if (sz >= copy_size) {
            break;
        }
    }
```

utzig, 2020-10-05:
> mcuboot reads 0xFFFFFFFF after swap was interrupted, because this is what was contained originally in app.hex

The trailer is only present in an image file if you used `--pad`, but even so the trailer itself is not part of the image; the bootloader always erases the trailer area when starting a new swap, and then writes metadata there. I always felt a bit worried about this particular `while (1)` you mentioned but I think there might be some other problem going on. Any chance you could get me a dump of the flash of the bricked device?

mniestroj, 2020-10-05:
Well, it didn't erase the end of first partition, because there are magic bytes and image_ok on the end (just like generated with `--pad --confirm`). Flash content was read using `nrfjprog --readcode /tmp/flash.hex`, gzipped and attached here: [flash.hex.gz](https://github.com/JuulLabs-OSS/mcuboot/files/5330060/flash.hex.gz).

utzig, 2020-10-05:
> Well, it didn't erase the end of first partition, because there are magic bytes and image_ok on the end

Well it does also write back this information once a swap upgrade has finished. What is your partition map?

mniestroj, 2020-10-05:
>Well it does also write back this information once a swap upgrade has finished. What is your partition map?
I am pretty sure trailer of first partition was not modified (yet), because when debugging I was looking for `boot_write_magic()` being called, and that didn't happen.

Partition map is stock from `nrf52840dk_nrf52840` Zephyr board: https://github.com/zephyrproject-rtos/zephyr/blob/2d7ccb59a9d3b501229e52f3369af8a1c45f2192/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts.

utzig, 2020-10-06:
> I am pretty sure trailer of first partition was not modified (yet), because when debugging I was looking for `boot_write_magic()` being called, and that didn't happen.

It only happens before a swap starts, not at every swap operation or reboot, and your swap status area shows there was a swap under way, but the swap info/size metadata was not written for some reason, which is why the bootloader does not know what to do. Are you able to reproduce this when starting from a freshly erased device?

> Partition map is stock from `nrf52840dk_nrf52840` Zephyr board: https://github.com/zephyrproject-rtos/zephyr/blob/2d7ccb59a9d3b501229e52f3369af8a1c45f2192/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts.

Is your device config also stock from `nrf52840dk_nrf52840` or have you customized something?

mniestroj, 2020-10-06:
> It only happens before a swap starts, not at every swap operation or reboot, and your swap status area shows there was a swap under way, but the swap info/size metadata was not written for some [...]

From what I understand, there are two phases: BOOT_STATUS_OP_MOVE and BOOT_STATUS_OP_SWAP. I think that we are still in BOOT_STATUS_OP_MOVE when rebooting. Yes, I am able to reproduce from freshly erased device.

>Is your device config also stock from nrf52840dk_nrf52840 or have you customized something?

Stock. Everything is described in "To Reproduce" at https://github.com/zephyrproject-rtos/zephyr/issues/28933. There is every command and action needed (including after which period (2s) reset was issued after mcuboot started).

PR #830 (2020-10-07): bootutil: fix swap-move brick with padded image in primary slot
When the image in the primary slot is padded, the boot source is considered the primary slot; this results in skipping the typical initialization of the trailer, which ends up bricking the device. As it is fine to always initialize the trailer in the primary slot when starting a new upgrade the extra check was removed.

Should fix #829 when done.
closed 2020-10-07 by utzig