mcu-tools/mcuboot #833: zephyr: disable HW stack protection before run application

nvlsianpu, 2020-10-07:
Zephyr might enable HW stack protection by default, which
propagates in mcuboot build. Bootloader works fine with that option, but
the problem arises when we jump to application code.

HW stack protection catches application code during early init, which results in
"unresponsive" application.

So fare issue is hidden thanks to #828 hack

- [ ] support ARM leverages SPLIM devices (Cortex M33...)
- [ ] support ARM leverages MPU devices

mniestroj, 2020-10-07:
I think we should have some generic API for disabling HW stack protection, possibly in Zephyr, then use it in mcuboot. Implementation details such as that implemented in #824 should be hidden in `arch/` directory.

nvlsianpu, 2020-11-03:
__weak void z_arm_clear_arm_mpu_config(void) from https://github.com/mcu-tools/mcuboot/pull/847 supports ARM leverages MPU devices. Also zephyr introduces https://github.com/zephyrproject-rtos/zephyr/pull/29400 so mcuboot won't need to cleanup register if application will use this functionality.

closed 2020-11-03 by nvlsianpu