mcu-tools/mcuboot #837: MCUBoot : encryption code issue

BohdanKovalchuk, 2020-10-08:
The file loader.c has an API boot_copy_region() (line  768)

boot_copy_region() has following code under if (IS_ENCRYPTED(hdr)) :

- uint8_t buf[1024]; declaration
- idx = hdr->ih_hdr_size; assignment
- boot_encrypt( .... , &buf[idx]); 

In case of header_size is 1024 last line will cause a RAM corruption.

A header size of 1024 as easiest way to satisfy vector table alignment (256 x 32bit) for Cortex M on Cypress PSoC6.

BohdanKovalchuk, 2020-10-09:
obviously, hotfix can be a declaration of larger buffer:

uint8_t buf[2048];

utzig, 2020-10-09:
Makes sense, in practice this should affect flash devices which have alignments larger than 512, which we don't support yet. Or someone could decide to have a very big header size as well, but for no reason. I think a proper fix is not to enlarge the buffer, but to update the code to deal with sizes larger than the buffer. Feel free to work on a fix or assign to me.

closed 2021-06-03 by github-actions[bot]