mcu-tools/mcuboot #839: Consider development designation for version information

nvlsianpu, 2020-10-14:
Readme.md stands that the version is the latest released version: `This is mcuboot version 1.6.0`

This is quite misleading on the first glance. May we distinguished development versions from any release versions?

solutions:
After each release change version info so it will reflect that it is development version.

1. use patch-level version equal to `99` field to mark development version like in zephyr:
last release: 1.6.0 -> development after that release: 1.6.99
next tag might be: 1.6.1 , 1.7.0 or 2.0.0

2. use `dev` suffix:
next major release: 1.7.0  -> development before that release: 1.7.0-dev
assuming that last release was 1.6.0 then next tag might be: 1.6.1 , 1.7.0 or 2.0.0

nvlsianpu, 2020-10-14:
cc @rlubos

d3zd3z, 2020-10-15:
I prefer the -dev approach to the .99 approach, mostly because the 99 seems kind of arbitrary. I'm trying to read through the semantic versioning spec to figure if there is an official way to specify a dev version.

Overall, though, I think this is a good idea, even though it introduces some changes.

One thought, maybe the releases should be done on a branch, and the master branch never leaves the -dev designation, and it is only removed in the release branch.

d3zd3z, 2020-10-15:
So, semantic versioning considers the pre-release designator in alphabetical order, which means 'dev' would come after 'alpha', 'beta', etc. This might be ok, since we don't actually use these (and we aren't ever actually releasing a 'dev' version). The 'rc' versions would come after this, so it might be ok to use.

d3zd3z, 2020-10-27:
Per team discussion. We will change the version to 1.8.0-dev immediately after the 1.7.0 release. Need to update release documentation to describe this.

PR #859 (2020-10-30): doc/release: Describe development version designation
After each release version will be changed to
MAJOR.MINOR.PATCH-dev.

fixes #839
closed 2020-10-30 by nvlsianpu