mcu-tools/mcuboot #849: New feature request: Common secondary slot for all images

cy-vaira, 2020-10-23:
MCUboot by design requires a pair of slots (primary and secondary) for every image and it can handle update of all the images at once after the reset. The secondary slots are unused except during the FW update. With this scheme, a multi-image application would require a lot of memory (flash) that would go unused most of the time application is running. 

Some applications may be looking to upgrade one image at a time in order to maximize the memory usage. This is in a way possible by configuring the flash map such that the secondary slot for all the images point to the same memory area. But this alone is not sufficient. Application should be able to tell MCUboot which image (i.e. image 1 or 2 or 3) is currently in the common secondary slot that it has to upgrade.

utzig, 2020-10-23:
/cc @tamasban @davidvincze

tamasban, 2020-10-26:
I agree that this solution is very desirable from flash utilization point of view.

My thoughts on the topic:

- Might header or TLV can convey the info, that which image must be updated with the image in the common secondary slot.
- I can see an issue with dependent multiple image update scenario. If two image must be updated the same time, because of API dependency or on armv8m platform: veneer region is relocated in secure image. As I see in this case both image must be present at the same time on the device to be able to perform an atomic update. Solution could be to have a special purpose image to handle this scenario which only responsible to download new images. Then this image can be started in the middle of the "atomic" update instead the main image. But it takes flash space as well so unsure whether finally it would add any gain.

cy-vaira, 2020-10-29:
Should we consider not supporting "dependent image update" when common secondary slot feature is enabled? 

In this case, the dependency is managed or ensured by the main app (that implements the OTA update capability). i.e. It's the responsibility of the main app to ensure all the latest images have been downloaded & updated (by contacting the update server more than once).

cy-vaira, 2020-10-29:
cc @romanjoe @BohdanKovalchuk

closed 2021-06-03 by github-actions[bot]