mcu-tools/mcuboot #850: imgtool.py cannot encrypt with X22619

ryangenz, 2020-10-23:
I have successfully encrypted .hex images with different types of .pem files.

But it is failing to encrypt with X22619 format. 
The error is seems to be:
  File "/usr/local/lib/python3.8/site-packages/imgtool/keys/x25519.py", line 100, in sign_digest
    return self.key.sign(data=digest)
AttributeError: '_X25519PrivateKey' object has no attribute 'sign'

utzig, 2020-10-23:
> AttributeError: '_X25519PrivateKey' object has no attribute 'sign'

A X25519 key cannot be used to sign, only encrypt, maybe you're trying to use it for both things?

ryangenz, 2020-10-23:
Yes,I am encrypting an image file. 

I am using it in an identical manner to ED25519 which succeeds. 
For example, this succeeds:
west sign -t imgtool -- --key ../../../../MyKey-ED25519.pem
And this fails:
west sign -t imgtool -- --key ../../../../MyKey-X25519.pem

(I have also tested with the full path and options for imgtool.py, instead of through west, but the results are the same)

utzig, 2020-10-23:
That is not how you encrypt an image, read this: https://mcuboot.com/mcuboot/imgtool.html#signing-images

What you want to do is: `west sign -t imgtool -- --key ../../../../MyKey-ED25519.pem -E ../../../../MyKey-X25519.pem`

ryangenz, 2020-10-23:
Yes, that worked for encryption.
Am I correct to understand that handling signing and encryption with X25519 is rather different than other encryption types?

This extra step has not been needed with the other types, and they are all working perfectly.

utzig, 2020-10-23:
> Am I correct to understand that handling signing and encryption with X25519 is rather different than other encryption types?
> 
> This extra step has not been needed with the other types, and they are all working perfectly.

The difference with curve25519 is that signing requires a ed25519 key and encryption a x25519 key. For RSA and secp256r1 you can use the same key type for both operations, but if you are **actually using the same key for both purposes** you are probably just giving up on security.

closed 2021-06-03 by github-actions[bot]