mcu-tools/mcuboot #858: Building with Zephyr fails on frdm_k64f (maybe others?)

utzig, 2020-10-30:
This is a Zephyr issue but I will open here because it happens with MCUboot upstream. When building the frdm_k64f board I am currently getting this error:

```
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c:24:7: error: expected ';' before 'void'
   24 | __weak void z_arm_clear_arm_mpu_config(void)
      |       ^~~~~
      |       ;
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c: In function 'z_arm_clear_arm_mpu_config':
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c:29:5: error: 'MPU' undeclared (first use in this function); did you mean 'FPU'?
   29 |   ((MPU->TYPE & MPU_TYPE_DREGION_Msk) >> MPU_TYPE_DREGION_Pos);
      |     ^~~
      |     FPU
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c:29:5: note: each undeclared identifier is reported only once for each function it appears in
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c:29:17: error: 'MPU_TYPE_DREGION_Msk' undeclared (first use in this function)
   29 |   ((MPU->TYPE & MPU_TYPE_DREGION_Msk) >> MPU_TYPE_DREGION_Pos);
      |                 ^~~~~~~~~~~~~~~~~~~~
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c:29:42: error: 'MPU_TYPE_DREGION_Pos' undeclared (first use in this function)
   29 |   ((MPU->TYPE & MPU_TYPE_DREGION_Msk) >> MPU_TYPE_DREGION_Pos);
      |                                          ^~~~~~~~~~~~~~~~~~~~
/home/utzig/src/embedded/zephyrproject/bootloader/mcuboot/boot/zephyr/arm_cleanup.c:32:3: warning: implicit declaration of function 'ARM_MPU_ClrRegion' [-Wimplicit-function-declaration]
   32 |   ARM_MPU_ClrRegion(i);
      |   ^~~~~~~~~~~~~~~~~
[46/240] Building C object zephyr/CMakeFiles/zephyr.dir/lib/os/heap-validate.c.obj
```

The build runs clean on `nrf52840dk_nrf52840`, so it seems some headers are missing for frdm_k64f board/soc/etc. And maybe for other boards as well?

/cc @nvlsianpu @ioannisg

nvlsianpu, 2020-10-30:
This was fixed by https://github.com/mcu-tools/mcuboot/pull/857

utzig, 2020-10-30:
I am running current Zephyr `master` (0bcd66ac36d8c02d2c3b3851c4086e8b692578a6) and still have the same issue

closed 2020-10-30 by utzig