mcu-tools/mcuboot #873: Useless test in LOAD_IMAGE_DATA

TTornblom, 2020-11-11:
Just noticed that there is a test on the return value from memcpy() in the LOAD_IMAGE_DATA() macro where the return value is compared to the first argument to memcpy().

This is a useless test as memcpy() always return its first argument, there is no other return value defined for this function.

utzig, 2020-11-11:
What would you suggest do be done instead?

TTornblom, 2020-11-11:
I'm not sure.

If MCUBOOT_RAM_LOAD is not defined, then it will use flash_area_read() instead of memcpy(), and I assume that will return some kind of error status if it fails, but if it is defined, then this test will always fail, so perhaps something like:

`#define LOAD_IMAGE_DATA(hdr, fap, start, output, size)       \
    (memcpy((output),(void*)((hdr)->ih_load_addr + (start)), \
            (size)), 0)
`

This will give a compiler a better chance of eliminating the dead code

tamasban, 2020-11-12:
Thanks @TTornblom for the proposal! Tbh never seen such a syntax, but it works with ARMCLANG and GNUARM as well. It spares around 50bytes of code. I will test it on Musca-A, and open PR later if works.

davidvincze, 2020-11-12:
I have never seen anything like this before: `rc = (memcpy(...), 0);`
Nor was I sure the syntax is valid. :D

utzig, 2020-11-12:
> Nor was I sure the syntax is valid. :D Does it have a name?

```
6.5.17 Comma operator

Syntax

1 expression:
       assignment-expression
            expression , assignment-expression

Semantics

2 The left operand of a comma operator is evaluated as a void expression; there is a sequence point
between its evaluation and that of the right operand. Then the right operand is evaluated; the result
has its type and value.114)
```

Source: http://www.open-std.org/jtc1/sc22/WG14/www/docs/n1570.pdf

tamasban, 2020-11-12:
https://github.com/mcu-tools/mcuboot/pull/877

Tested on Musca-A with GNUARM.

davidvincze, 2020-11-24:
PR#877 has been merged which provides a fix, so I'm closing this issue.

closed 2020-11-24 by davidvincze