mcu-tools/mcuboot #905: mcuboot corrupts zephyr application data

AnanthJaisankar, 2021-01-05:
i have some application variables declared as __noinit which should be retained in system off mode(nrf52840). I have tested without mcuboot bootloader and it is working but not working with bootloader. I am not sure but mcuboot clearing it off somewhere or do i need to enable some configs to avoid this problem?
Thanks in advance.

d3zd3z, 2021-01-12:
MCUboot currently assumes that all RAM is available while it is running, so these values would not be preserved. In order to preserve values that you need, the linker scripts in both the application and mcuboot would need to be enhanced to place these values in some memory that wouldn't then be overwritten by MCUboot.

jameswalmsley, 2021-01-27:
In your zephyr board layer you can add a custome-sections.ld file and then create a section for your noinit variables:

```
my_noload_data : (NOLOAD): {
  *(.my_info)
}
```

Then in your source code mark the variables you don't want to be replaced with:

```
static struct myinfo info __attribute__((section(".my_info")));
```

AnanthJaisankar, 2021-01-28:
Thanks for the responses. I will try this and let you know.

jameswalmsley, 2021-01-28:
One other thing, I forgot, you have to specify the address where you
want that section to be placed. (Otherwise the bootloader/application
will be linked differently).

The full definition of the linker section looks like:

```
bootloader_info 0x20000000 (NOLOAD): {
  *(.bootinfo)
}
```

You may want to define a structure with some kind of fixed size /
reserved space. (You can also specify that in the linker too).

Ensure that the structure remains compatible between versions of your
bootloader and application.

E.g. if you double the size of your structure in the future, and update
a device with the old bootloader, then the bootloader will only know not
to initialise the first half.

d3zd3z, 2021-01-29:
Also, keep in mind that values allocated in this section will not be initialized (otherwise, data wouldn't be preserved). You'll probably want to have some kind of magic value in this section to indicate that any of the values should be considered meaningful, and if that isn't set, clear the entire section.

closed 2021-06-03 by github-actions[bot]