mcu-tools/mcuboot #912: Create v1.7.x branch

d3zd3z, 2021-01-12:
These are the current changes in the main branch that are not in 1.7.0. I've checked what I think is appropriate for 1.7.1.

- [ ] ac61c2e4 Fix nokogiri<=1.11.0.rc4 vulnerability
- [ ] 6907c90e ci: pull trusted-firmware-m repo on Travis run
- [ ] bd0ce620 ci: pull fih-test docker image on install
- [ ] a069befe ci: docker: remove source repos from fih-test
- [ ] 48a4ec3e ptest: Use github workflow instead of travis
- [ ] ae8f8db7 ptest: Update dependencies
- [ ] f6847381 ci: Pin tf-m-tests version in FIH test
- [ ] 008f4a7a boot: zephyr: cleanup NXP MPU configuration before boot
- [ ] 7cca88ab ci: Update tf-m version in FIH test
- [ ] 28779658 Enable support for building mcuboot for Mbed with direct-xip
- [ ] 50820b16 mgtool: Add support for setting fixed ROM address into image header
- [ ] 81d19f01 bootutil: Add cmake build file
- [ ] d1233e1b Add reference counting to Mbed OS flash backend
- [ ] ae097709 Fix Mbed OS logging port when trace is disabled.
- [ ] e75e33d9 boot: zephyr: Default to LOG_MINIMAL
- [ ] 83ec8426 zephyr/Kconfig: Added default pin for serial recovery mode for nRF5340DK
- [ ] efb32039 zephyr: use minimal CBPRINTF implementation
- [ ] 39b6518d Post v1.7.0 action
- [ ] 0f409b05 ext: tinycrypt: update ctr mode to stream
- [ ] d3838989 bootutil: crypto: avoid unuseful memset
- [ ] 1d0467e8 boot/mynewt: De-initialize peripherals before calling app

utzig, 2021-01-12:
I don't see the checks here, and I can changed them but I don't think they are saved on the server.

d3zd3z, 2021-01-12:
Do you see the one I made? I saw a check briefly appear and then go away, and it said you had edited it. I have only checked the 'mgtool' one so far.

utzig, 2021-01-12:
> I don't see the checks here, and I can change them but don't think they are saved on the server.

Now it did show up!

utzig, 2021-01-12:
@de-nordic Could you double check this is the commit you need? ^

d3zd3z, 2021-01-12:
I guess the main decision is what to include. The imgtool one is what is driving this, but what about the CI/ptest ones?

d3zd3z, 2021-01-12:
And maybe the ruby one to keep away vulnerability alerts from showing up.

utzig, 2021-01-12:
> And maybe the ruby one to keep away vulnerability alerts from showing up.

I assume the pages are only published from `master` so Jekyll would only run there. I have no idea if GH will show the vulnerability.

d3zd3z, 2021-01-13:
Resolved by #917.

closed 2021-01-21 by d3zd3z