mcu-tools/mcuboot #924: imgtool.py cannot verify the img with TLV protect info

PengXuu, 2021-01-21:
I use command as below to sign the image  with  TLV Prot info：security count：
`imgtool.py sign -k root-rsa-2048.pem -e big -S 0x28000 -H 0x0400 --pad-header -v "1.0.0" -s auto --align 8 Project.bin test_sign.bin`
Then i use 'verify' command to verify the signed image, failed with log "Invalid TLV info magic; is this an MCUboot image?" 

After i check the function verify in image.py, it seem dont adapt to image include TLV_PROT_INFO_MAGIC
```
        tlv_info = b[header_size+img_size:header_size+img_size+TLV_INFO_SIZE]
        magic, tlv_tot = struct.unpack('HH', tlv_info)
        if magic != TLV_INFO_MAGIC:
            return VerifyResult.INVALID_TLV_INFO_MAGIC, None, None

```

utzig, 2021-01-21:
I can confirm it has the issue you mentioned, and it also does not seem to support big endian.

utzig, 2021-01-21:
The issues should be fixed by #926, please confirm if possible.

PengXuu, 2021-01-25:
Thanks for your answer.
I confirmed that this issue was solved by #926

closed 2021-01-25 by PengXuu