mcu-tools/mcuboot #929: Support external crypto unit

Formatsh, 2021-01-25:
Hi,
I'm working on a device that uses Zephyr + mcuboot on nRF52840. Since flash protection on this (and almost all other non-secure microcontrollers) is broken, I'd like to use external secure key storage / accelerator to verify firmware signature and to decrypt it.

I have an ATECC608B on board, and due to documentation restrictions and dependency on cryptoauthlib, it's not possible to commit code directly using it. But it would be nice to create a compile option+  API, that would allow:
- external API for signing (low priority)
- external API for decryption (high priority, storing private key in generic MCU is not safe)

This would than allow anyone linking their own decryption code without a need to store private keys in insecure flash.

Would there be an interest in accepting such patch?

nvlsianpu, 2021-01-25:
Are you sure about that you want to store a private key for decryption?

Formatsh, 2021-01-26:
Well at least that's how zephyr bootloader variant works. You sign the image with private key and encrypt the image with public key of different pair.
Going on examples/help available for imgtool, it looks like this:

```
west sign -t imgtool -d fw_build/ -- --key ~/secret/project_sign_dev_priv.pem -E ~/secret/project/_enc_dev_pub.pem -v 1.0.0+100
```

And for it to work, you have to store your private key in **mcuboot/boot/zephyr/keys.c**
See https://github.com/mcu-tools/mcuboot/blob/master/boot/zephyr/keys.c

Now I know it generates ephemeral key which is used to encrypt each firmware file separately, but if the private key is stored somewhere in flash, it is kind of moot point, isn't it?

Unless I'm completely misunderstanding something :)

utzig, 2021-01-26:
> Now I know it generates ephemeral key which is used to encrypt each firmware file separately, but if the private key is stored somewhere in flash, it is kind of moot point, isn't it?

Yes, if flash access is available (eg, JTAG/SWD) anyone can recover the decryption key. Also the image in flash is not encrypted while in slot 0 (image that runs), so recovering the key or recovering the image itself, which is what is protected in the first place, are equivalent. This was meant to protected in transit while data is sent to the device.

Formatsh, 2021-01-26:
Yes, I'm aware firmware in slot 0 is not encrypted. JTAG/SWD will be disabled on production units, but since attacks like these exists: https://limitedresults.com/2020/06/nrf52-debug-resurrection-approtect-bypass/, the keys are not safe even with JTAG disabled.

The subtle difference is only that by recovering key from flash, you're able to decrypt any future firmware without having to dump it from device. That makes it easier to make counterfeit devices.

I'll probably try to implement it if I have time and you can then decide whether it's interesting or not.

closed 2021-06-03 by github-actions[bot]