mcu-tools/mcuboot #930: License headers and SPDX identifiers missing

jainvikas8, 2021-01-25:
I have comes across License header and SPDX identifiers missing on most of these https://github.com/mcu-tools/mcuboot/tree/master/scripts

Wonder if this can be fixed, please?

closed 2021-01-28 by nvlsianpu

jainvikas8, 2021-01-28:
I don't think #933 fixes both the issues here.

Copyright/License header is still missing, please that be added to?

d3zd3z, 2021-01-29:
What are you finding still missing? In general, most open source projects have determined that an SPDX license is sufficient as a license marking. As far as copyright markings, I'm not really able to add those for entities other than my own employee, so if people feel they need to add them, they would have to do that themselves.

PR #933 (2021-01-28): scripts: Add SPDX headers
Add SPDX headers indicating the Apache-2.0 license to the scripts in the
scripts directory.  This can be assumed due to the presence of the
Apache-2.0 LICENSE file at the top of the project.

Fixes #930
commit 79c4fcf4 (2021-01-28): scripts: Add SPDX headers

Add SPDX headers indicating the Apache-2.0 license to the scripts in the
scripts directory.  This can be assumed due to the presence of the
Apache-2.0 LICENSE file at the top of the project.

Fixes #930