mcu-tools/mcuboot #940: Particle Xenon is unable to start bootloader

DuMaM, 2021-02-04:
I was trying to work with mcuboot for zephyr without luck. 

Version that was able to compile was:
v1.7.0 mcuboot (master do not work)
v2.4.0 zephyr
zephyr-sdk-0.12.2

Then i programed it with opendocd;
openocd -f interface/ftdi/jtag-lock-pick_tiny_2.cfg  -c "transport select swd" -f target/nrf52.cfg -c "program zephyr/zephyr.elf verify reset exit"

And app no response for bootloader.

d3zd3z, 2021-02-09:
Are you able to build the Zephyr app `samples/hello_world` for your target, and do you get the printed messages. MCUboot is basically just a Zephyr app, so it'd be good to make sure this works.

DuMaM, 2021-02-14:
building and flashing mcutool
 ``` bash
 HEAD is now at e757f7b Version updates for 1.7.1
 ~/A/z/m/b/z/build  ➦ e757f7b   cmake -GNinja -DBOARD=particle_xenon ..
Including boilerplate (Zephyr base (cached)): /home/dumam/Apps/zmk/zmk/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/dumam/Apps/zmk/mcuboot/boot/zephyr
-- Zephyr version: 2.4.99 (/home/dumam/Apps/zmk/zmk/zephyr)
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: particle_xenon
-- Cache files will be written to: /home/dumam/.cache/zephyr
-- Found toolchain: zephyr (/home/dumam/zephyr-sdk-0.12.2)
-- Found dtc: /home/dumam/zephyr-sdk-0.12.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/dumam/Apps/zmk/zmk/zephyr/boards/arm/particle_xenon/particle_xenon.dts
-- Found devicetree overlay: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/dts.overlay
-- Found devicetree overlay: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/dumam/Apps/zmk/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/dumam/Apps/zmk/zmk/zephyr/boards/arm/particle_xenon/particle_xenon_defconfig'
Merged configuration '/home/dumam/Apps/zmk/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/dumam/zephyr-sdk-0.12.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /home/dumam/Apps/zmk/zmk/zephyr/kernel/CMakeLists.txt:63 (message):
  Single threaded mode (CONFIG_MULTITHREADING=n) is deprecated

MCUBoot bootloader key file: /home/dumam/Apps/zmk/mcuboot/root-rsa-2048.pem
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build
 ~/A/z/m/b/z/build  ➦ e757f7b   ninja
[1/245] Preparing syscall dependency handling

[240/245] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       43424 B        48 KB     88.35%
            SRAM:       22544 B       256 KB      8.60%
        IDT_LIST:          72 B         2 KB      3.52%
[245/245] Linking C executable zephyr/zephyr.elf
 ~/A/z/m/b/z/build  ➦ e757f7b   openocd -f interface/ftdi/jtag-lock-pick_tiny_2.cfg  -c "transport select swd" -f target/nrf52.cfg -c "program zephyr/zephyr.elf verify reset exit"
Open On-Chip Debugger 0.10.0+dev-snapshot (2020-08-19-11:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : FTDI SWD mode enabled
swd
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00002714 msp: 0x20005010
** Programming Started **
Info : nRF52840-xxAA(build code: C0) 1024kB Flash, 256kB RAM
Warn : Flash protection of this nRF device is not supported
Warn : Adding extra erase range, 0x0000a9a0 .. 0x0000afff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
 ~/A/z/m/b/z/build  ➦ e757f7b   
```
 
 sample/hello_world 
``` bash
openocd -f interface/ftdi/jtag-lock-pick_tiny_2.cfg  -c "transport select swd" -f target/nrf52.cfg -c "program zephyr/zephyr.elf verify reset exit"
Open On-Chip Debugger 0.10.0+dev-snapshot (2020-08-19-11:14)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : FTDI SWD mode enabled
swd
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00002714 msp: 0x20005010
** Programming Started **
Info : nRF52840-xxAA(build code: C0) 1024kB Flash, 256kB RAM
Warn : Flash protection of this nRF device is not supported
Warn : Adding extra erase range, 0x00003888 .. 0x00003fff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked
```

DuMaM, 2021-02-15:
I updated zephyr project with `west update` and I was able to build mcuboot from master. 
Still do not work for sample/hello world.
mcugit: ``` bash
commit 8a5e49876eae4f2d9bc8ced2fa2e2be0d24ddd68 (HEAD -> master, origin/master, origin/HEAD)
Author: Fabio Utzig <utzig@apache.org>
Date:   Thu Feb 11 19:46:58 2021 -0300

    imgtool: rollback to 1.7.0 release
    
    Since there won't be anymore alpha releases, rollback to latest official
    (from master). The next released versions now can be either 1.7.X from
    v1.7-branch or X.Y.Zrc1 for the release candidate of next stable
    version.
    
    Signed-off-by: Fabio Utzig <utzig@apache.org>

diff --git a/scripts/imgtool/__init__.py b/scripts/imgtool/__init__.py
index 5d7d181..4f19399 100644
--- a/scripts/imgtool/__init__.py
+++ b/scripts/imgtool/__init__.py
@@ -14,4 +14,4 @@
 # See the License for the specific language governing permissions and
 # limitations under the License.
 
-imgtool_version = "1.8.0a1"
+imgtool_version = "1.7.0"
``` 

``` bash
 ~/A/z/m/b/z/build   master  cmake -GNinja -DBOARD=particle_xenon ..
Including boilerplate (Zephyr workspace): /home/dumam/Apps/zmk/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/dumam/Apps/zmk/mcuboot/boot/zephyr
-- Zephyr version: 2.5.99 (/home/dumam/Apps/zmk/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.6") found components: Interpreter 
-- Found west (found suitable version "0.9.0", minimum required is "0.7.1")
-- Board: particle_xenon
-- Cache files will be written to: /home/dumam/.cache/zephyr
-- Using toolchain: zephyr 0.12.2 (/home/dumam/zephyr-sdk-0.12.2)
-- Found dtc: /home/dumam/zephyr-sdk-0.12.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/dumam/Apps/zmk/zephyr/boards/arm/particle_xenon/particle_xenon.dts
-- Found devicetree overlay: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/dts.overlay
-- Generated zephyr.dts: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/include/generated/device_extern.h
Parsing /home/dumam/Apps/zmk/mcuboot/boot/zephyr/Kconfig
Loaded configuration '/home/dumam/Apps/zmk/zephyr/boards/arm/particle_xenon/particle_xenon_defconfig'
Merged configuration '/home/dumam/Apps/zmk/mcuboot/boot/zephyr/prj.conf'
Configuration saved to '/home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/dumam/Apps/zmk/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/dumam/zephyr-sdk-0.12.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
CMake Warning at /home/dumam/Apps/zmk/zephyr/kernel/CMakeLists.txt:59 (message):
  Single threaded mode (CONFIG_MULTITHREADING=n) is deprecated

MCUBoot bootloader key file: /home/dumam/Apps/zmk/mcuboot/root-rsa-2048.pem
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dumam/Apps/zmk/mcuboot/boot/zephyr/build
```

d3zd3z, 2021-02-19:
In case it is hard to find elsewhere, basically releases of Zephyr will have a corresponding version of mcuboot, generally that is maintained in the west module for mcuboot. It should also be the case that master on mcuboot and master on Zephyr should work together. This does sometimes break, but it is reasonable to bring that up (or file an issue). It looks like this is currently (2021-02-19) an issue.

My first suggestion on the hello_world was to try building hello world without mcuboot at all. Just load it as the only application on the target, and make sure that the board support for that target is working. That should also tell us if the printk messages on Zephyr are making it somewhere. I don't know if this board has a UART or some other method to debug. Once that is working we can then try to figure out where mcuboot isn't working, since MCUboot itself is kind of like an ordinary Zephyr application.

DuMaM, 2021-02-20:
@d3zd3z Thanks, I managed to exec Helloworld and other samples. :tada:  :+1: 
There was long time since I was developing on other stuff than RPi, and I forgot that console is not always in USB output. :disappointed:. When I connected UART converter to proper ports I got output which help me a lot. 
I think that right now I will manage to solve mcuboot problem by myself.

d3zd3z, 2021-02-22:
Feel free to continue to ask questions, either here, or on slack. Sometimes debugging issues with mcuboot startup can be a bit tricky, because of the pickiness of signature verification.

DuMaM, 2021-02-22:
I would rather contact on slack. :+1: Thanks for help :)

closed 2021-02-22 by DuMaM