mcu-tools/mcuboot #942: Add mcuboot_app and blinky_app for Texas Instruments CC1352R1

tcave6677, 2021-02-08:
Create mcuboot_app and blinky_app projects for the CC1352R1 LaunchPad:
https://www.ti.com/tool/LAUNCHXL-CC1352R1

The TI NoRTOS DPL (Driver Porting Layer) from the simplelink_cc13x2_26x2_sdk_4_40_xx_xx SDK (https://www.ti.com/tool/SIMPLELINK-CC13X2-26X2-SDK) should be used so that the SimpleLink Crypto Driver is available for HW Crypto support in future.

The initial port only requires support for XIP mode which ‘ping / pongs’ between running the latest FW version from the Primary or Secondary Slot. No security is required at this stage, however if security support is required for bootutil to build then TinyCrypt should be included.

No serial boot application will be provided. Instead images will be loaded through the ROM’ed serial boot loader in the CC1352. A host tool shall be provided to load the images. The ROM’ed serial boot loader runs at startup before running the mcuboot_app. If the pin configured as the boot loader backdoor is asserted at startup then the ROM’ed boot loader will be invoked and wait for a new mage to be loaded through the serial port. More information on the ROM'ed serial boot loader can be found in section 10 of the CC1352 TRM:
https://dev.ti.com/tirex/explore/node?node=AFAlDSad-dUrZQZxqwVljg__coGQ502__LATEST

tcave6677, 2021-02-08:
Hi, I'm adding support for CC1352 to MCU Boot on behalf of Texas Instruments. I have looked through the documentation on contributing (https://mcu-tools.github.io/mcuboot/SubmittingPatches.html). It looks like I should create a branch and make commits against a Jira ticket. Would that be `JIRA: MCUB-942`, or does this issue need raising in a different system?

utzig, 2021-02-08:
> Hi, I'm adding support for CC1352 to MCU Boot on behalf of Texas Instruments. I have looked through the documentation on contributing (https://mcu-tools.github.io/mcuboot/SubmittingPatches.html). It [...]

That document needs updating, you can ignore Jira and just submit a normal PR here on GH.

closed 2021-06-03 by github-actions[bot]