mcu-tools/mcuboot #944: Why mcuboot ram load not support  image encryption?

Cherish-forever, 2021-02-09:
I have read mcuboot design guide, I am not understand why ram load not support image encryption?
In image slot mode or direct-xip mode, there always have some method to attacks flash.
exp:
1. read all flash
2. write a boot image jump to image slot
3. merge boot and image slot as a compound binary
4. burn to MCU

image slot mode or direct-xip mode, the firmware is exposed.
if ram load support image encryption. the image in flash is encryption.
if need run firmware, mcuboot must decryption the flash and load to ram, even if firmware has been read , it is encryption, generate boot just jump is no use.

Cherish-forever, 2021-02-09:
In ram load mode, two slot mode need encryption, ram load can support multiple encryption, not only one.

d3zd3z, 2021-02-12:
Fairly simple reason: it hasn't been implemented yet. I believe someone is working on this, however.

closed 2021-06-03 by github-actions[bot]