mcu-tools/mcuboot #958: Higher consumption on nRF52832 when boot from mcuboot.

vbelloir, 2021-03-06:
Hi everybody,
This issue may be linked with #529 but I'm not sure.

I'm trying to improve consumption of PineTime watch, using:
- mcuboot with mynewt as bootloader: https://github.com/JF002/pinetime-mcuboot-bootloader
- infinitime as OS and apps: https://github.com/JF002/InfiniTime

Infinitime is based on FreeRTOS.

Consumption is higher when Infinitime is booted by mcuboot than when it is stand-alone. Difference is about 500µA.

As a example, I disabled watchdog in bootloader, and add a simple __WFE loop in main function of Infinitime.
Booting from mcuboot gives a consumption of 574µA, while booting in stand-alone configuration gives 170µA.

Resources may be used in mcuboot/mynewt and not de-initialized in mcuboot, nor in Infinitime.

Do you have any idea for me? How can I identify which resource is consuming power?

d3zd3z, 2021-03-10:
One suggestion, that may help before #529 is fixed, would be to disable the UART and logging from within MCUboot. At least see if that helps with the consumption.

vbelloir, 2021-03-10:
Hello @d3zd3z and thanks for your answer.
If I well understand I have to add
```
CONFIG_SERIAL=n
CONFIG_LOG=n
```

in `repos/mcuboot/boot/zephyr/prj.conf`

But If I do that, exactly the same mynewt.elf.hex file is generated.
I must miss something.

d3zd3z, 2021-03-10:
It might need to be
```
# CONFIG_SERIAL is not set
# CONFIG_LOG is not set
```
I'm not exactly sure when one is required over the explicit 'n'. You can always run a menuconfig build, and see what it writes in the zephyr/.config file in the build output directory.

utzig, 2021-03-10:
Mynewt does not use config files, this would be for Zephyr. Please gist the output of `newt target brief <your-target>` somewhere so I can take a look.

vbelloir, 2021-03-11:
@utzig 
Here is the result of `newt target config brief targets/nrf52_boot`

I hope this is what you ask for.
https://pastebin.com/3ZAuhVeQ

@d3zd3z I really don't know how to use menuconfig. I will ask to my friend DuckDuckGo and come back to you.

utzig, 2021-03-11:
> @utzig
> Here is the result of `newt target config brief targets/nrf52_boot`
> 
> I hope this is what you ask for.
> https://pastebin.com/3ZAuhVeQ

Ok, I'll take a look later. I would also recommend to update to a version that at least includes this commit: https://github.com/mcu-tools/mcuboot/commit/1d0467e8339f91b0dac9cb0bcedb131cdf7ac63d

> @d3zd3z I really don't know how to use menuconfig. I will ask to my friend DuckDuckGo and come back to you.

This is only for Zephyr, you are on Mynewt so you don't need to do this.

vbelloir, 2021-03-11:
I would say that my newt version is new enough :
```
19:33 $ newt version
Apache Newt 1.9.0-dev / 5da6e2f / 2021-02-21_22:34
```

utzig, 2021-03-11:
> I would say that my newt version is new enough :
> 
> ```
> 19:33 $ newt version
> Apache Newt 1.9.0-dev / 5da6e2f / 2021-02-21_22:34
> ```

The `newt` tool version is irrelevant here, what you need is a recent MCUboot. If you are using `0-dev` you can just `new upgrade` and get more recent releases; use `newt info` to print the `sha`'s of your current repositories.

vbelloir, 2021-03-11:
ok, sorry for that, I'm really new to mynewt, and may say lot of wrong thing.

Here is the project.yml file:
```
repository.apache-mynewt-core:
    type: github
    vers: 1.8.0
    user: apache
    repo: mynewt-core

repository.mcuboot:
    type: github
    vers: 1.5.0
    user: JuulLabs-OSS
    repo: mcuboot

```

and here is the sha1:
`mcuboot: 310af87f61779f0f1930921055f2b19921a1ef45, 1.5.0,`

so yes, the commit you are pointed seems to be introduced in 1.7.0, so I have to update.

vbelloir, 2021-03-11:
OK, I put 0-dev for both mcuboot and mynewt, and upgrade process succeed.
Now, I will try to build it and check if it still works.

vbelloir, 2021-03-15:
Hi @utzig 
sorry for the delay.
So, I can confirm that even with 0-dev, consumption is still higher with bootloader than without.
Did you have time to see my pastebined conf?

vbelloir, 2021-03-23:
Hi @utzig can I ask you a little bit more help, checking the pastebin?

utzig, 2021-03-24:
> Hi @utzig can I ask you a little bit more help, checking the pastebin?

It looks pretty standard, I believe you can disable `TIMER_0` because it is only used by BOOT_SERIAL. Have you tried to compare with the config of your app?

vbelloir, 2021-03-24:
Thanks for your answer.  
How should I do that? modifying a syscfg.yml somewhere, or with the `newt target config` command?

I found this post on Nordic forum.
https://devzone.nordicsemi.com/f/nordic-q-a/42145/spim-easydma-to-increase-power-consumption/163946#163946
Maybe I have to check that also.

nvlsianpu, 2021-03-25:
@vbelloir You might try config `CONFIG_MCUBOOT_NRF_CLEANUP_PERIPHERAL` implemented in NCS mcuboot fork

vbelloir, 2021-03-26:
It's really interested, and seems closed to my issue.
How can I do that? I should change mcuboot repo in project.yml to point to nrfconnect one?

nvlsianpu, 2021-03-30:
^^ You are using NRF, you can port the code enabled by this configuration.

closed 2021-06-14 by github-actions[bot]